theyosh/TerrariumPI #307: Unable to add new switch or sensor - any type

skipperpl, 2019-06-28:
Hello, I'm unable to add switch or sensor, in debug we can see:

`KeyError: 'hardwaretype'
2019-06-28 14:23:49,498 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2019-06-28 14:23:54,557 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python3.5/dist-packages/bottle.py", line 1742, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 46, in webserver_headers
    return fn(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 88, in wrapper
    return func(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 284, in __update_api_call
    result['ok'] = self.__terrariumEngine.set_config(path,postdata,request.files)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1193, in set_config
    update_ok = self.set_sensors_config(data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 726, in set_sensors_config
    self.__load_sensors(data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 182, in __load_sensors
    sensordata['hardwaretype'],
KeyError: 'hardwaretype'
`
for the sensor, and respectively for the power switch as well.

Cheers 
Pawel

theyosh, 2019-06-28:
Hmm, there was a small change in the code that could cause this. But I have changed that, so could you do a `git pull` and restart?

And then see if it works? Else I have to look into it a bit more. But I am able to add new sensors, so I would expect that the latest code should work.

fenyx13, 2019-06-28:
I had the same issue adding sensors, the git pull fixed it. Thank you.

skipperpl, 2019-06-28:
Okay, sensors are now being added, issue still exist on adding switches, the only difference is that, error message is thrown almost immediately after pressing submit button.
`Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 862, in _handle
    return route.call(**args)
  File "/usr/local/lib/python2.7/dist-packages/bottle.py", line 1742, in wrapper
    rv = callback(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 46, in webserver_headers
    return fn(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 88, in wrapper
    return func(*a, **ka)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 284, in __update_api_call
    result['ok'] = self.__terrariumEngine.set_config(path,postdata,request.files)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1190, in set_config
    update_ok = self.set_switches_config(data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 753, in set_switches_config
    self.__load_power_switches(data)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 289, in __load_power_switches
    power_switch_config['hardwaretype'],
KeyError: 'hardwaretype'
`

Regards
Pawel

skipperpl, 2019-06-28:
Update:
I'v added to terrariumpi.js same line you have added to "sensors"
$('select[name*="hardwaretype"]:disabled:last').attr("disabled", false);
to the switches (in the same // Add switch position) and all is working now,
I think it also should be added to the "doors" section.

Regards 
Pawel

theyosh, 2019-07-01:
Changed the code once more. Now it will also add a nice border around new sensors/switches/doors etc..
And saving should now also work for all

skipperpl, 2019-07-02:
Thanks, now all works perfectly for me, thanks ;)

closed 2019-07-03 by theyosh