theyosh/TerrariumPI #308: Missing Python Libraries

Natrixz, 2019-07-02:
Heya =)

In the newest install.sh your missing "icalendar" because that one's missing it also won't install icalevents.
So anyone using the calendar will have issue's after they reboot.

theyosh, 2019-07-02:
Hi,
that is strange. I just did a manual install of icalevents, and I will get icalendar for free with it :P :+1: 
```
pip install icalevents
Collecting icalevents
  Downloading https://files.pythonhosted.org/packages/76/65/bc2d16014d92a6cf9e55c49d5ec25510355c4e812ca5ea04a1a2b45c6647/icalevents-0.1.20.tar.gz
Collecting datetime (from icalevents)
  Downloading https://files.pythonhosted.org/packages/73/22/a5297f3a1f92468cc737f8ce7ba6e5f245fcfafeae810ba37bd1039ea01c/DateTime-4.3-py2.py3-none-any.whl (60kB)
    100% |████████████████████████████████| 61kB 466kB/s 
Collecting httplib2 (from icalevents)
  Downloading https://files.pythonhosted.org/packages/5c/f3/7206894743389a4f727b73e6df4da60c9ee3cbef3f5afd82814592eafa8b/httplib2-0.13.0.tar.gz (219kB)
    100% |████████████████████████████████| 225kB 501kB/s 
Collecting icalendar (from icalevents)
  Downloading https://files.pythonhosted.org/packages/05/00/af65fe57330c5b087ee497d3e412b03432a1823a46bac6900f956b9393a1/icalendar-4.0.3-py2.py3-none-any.whl (73kB)
    100% |████████████████████████████████| 81kB 194kB/s 
Collecting pytz (from icalevents)
  Downloading https://files.pythonhosted.org/packages/3d/73/fe30c2daaaa0713420d0382b16fbb761409f532c56bdcc514bf7b6262bb6/pytz-2019.1-py2.py3-none-any.whl (510kB)
    100% |████████████████████████████████| 512kB 166kB/s 
Requirement already satisfied: zope.interface in /usr/lib/python2.7/dist-packages (from datetime->icalevents)
Requirement already satisfied: python-dateutil in /usr/local/lib/python2.7/dist-packages (from icalendar->icalevents)
Requirement already satisfied: six>=1.5 in /usr/local/lib/python2.7/dist-packages (from python-dateutil->icalendar->icalevents)
Building wheels for collected packages: icalevents, httplib2
  Running setup.py bdist_wheel for icalevents ... done
  Stored in directory: /root/.cache/pip/wheels/73/82/70/0a60a6ed76e057b623aa0c4f2b4f3083a35679b4af74cab4dd
  Running setup.py bdist_wheel for httplib2 ... done
  Stored in directory: /root/.cache/pip/wheels/e4/e5/3f/f7c5c820c180fdcf5b351116b1425f40fa07167e6f98a5310e
Successfully built icalevents httplib2
Installing collected packages: pytz, datetime, httplib2, icalendar, icalevents
Successfully installed datetime-4.3 httplib2-0.13.0 icalendar-4.0.3 icalevents-0.1.20 pytz-2019.1
```

But the icalevents package says it is only working for Python3, but I have just installed for python 2.7 . So maybe the pip3 is working different then pip2? Not sure...

So, for now I just add icalendar to the pip installer just to me sure. But I had expected that the dependencies should do the trick.

theyosh, 2019-07-05:
But I have added it explicitly so that should take care of. Thanks for spotting!

closed 2019-07-05 by theyosh