theyosh/TerrariumPI #309: Script-triggering switch

jornobe, 2019-07-02:
Now that we have a sensor type that triggers a script, would it be possible to have the same for a switch?

It was mentioned some time ago and it would be ideal for me to trigger my automatic feeder when out of the house for a while.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

theyosh, 2019-07-02:
Why can't that be done with the current power switches? Is there something that special that it cannot be done directly? Maybe I am missing something

jornobe, 2019-07-03:
I have built a device that can drop morio worms into a dish by running a script that spins a stepper motor. The script will first switch on the Energenie outlet the motor is connected to and disables it after the action as well. This way the motor doesn’t heat up when “idle” (steppers keep the coils powered to maintain position which generates a ton of heat). 

Having a power switch on the dashboard that triggers the script will take away the need to ssh into the device to run it. 

Some pictures of the feeding device:
[image]

Video of the device doing a full 8 cycle run:
https://1drv.ms/v/s!AloTfj3l56875BX-9OyP0GfcVqZO

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

theyosh, 2019-07-04:
Yes, that is totally clear. This 'trick' can't be done with my software. So I will add a script option switch.

The script will be started with a numeric value from 0 (off) to 100 (on). Because then I can also support dimming with scripts.

But it will still be a manual action, due to the fact I do not have a time schedule to run a power switch for X secs at moment XX:YY during a day. But first step is make it possible with the power switch through the web interface.

theyosh, 2019-07-05:
So, I added the new power switch script option. You can just enter any script in the address bar now and select for custom script type power switch.

When put on you will get a value of 100 as first argument. When off the value will be 0.

There is also an example script at the contrib folder. There I made the script switch_toggle.py that just outputs on and off. But there you can see how I read the argument values

jornobe, 2019-07-15:
Thanks a lot! Appreciate this addition!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

theyosh, 2019-07-28:
Closing due to implemented

closed 2019-07-28 by theyosh