theyosh/TerrariumPI #311: HTU21D does not work

swekley, 2019-07-08:
I just tried adding a HTU21D sensor, but it somehow doesn't work.

`load_raw_data temp:
'terrariumHTU21DSensor' object has no attribute '_terrariumI2CSensor__device_number'
load_raw_data humid:
'terrariumHTU21DSensor' object has no attribute '_terrariumI2CSensor__device_number'
2019-07-08 15:10:20,644 - WARNING - terrariumSensor      - Measured value NoneC from htu21d sensor 'htu' is outside valid range 0.00C - 0.00C in 0.19374 seconds.
2019-07-08 15:10:30,887 - WARNING - terrariumSensor      - Measured value NoneC from dht22 sensor 'dht22' is outside valid range 0.00C - 40.00C in 10.10805 seconds.`

I've tried using 0x40 as address instead of just 40 but then it says this 
`2019-07-08 15:09:51,575 - WARNING - terrariumI2CSensor   - Error opening htu21d sensor 'htu'. Error message: invalid literal for int() with base 16: '0x0x40'`

So that's not the problem, the sensor is also not the problem as it works without terrariumpi
`pi@raspberrypi:~/rpi-examples/HTU21D/c $ sudo ./HTU21D_test 
21.09C
54.10%rh`

theyosh, 2019-07-08:
hmm not sure, but I made a small change. Could you do a git pull and restart?

swekley, 2019-07-09:
Humidity works now, but temperature is stuck at 21.37 degrees and gives an error. can't seem to find anything in the console.

swekley, 2019-07-09:
a reboot fixed it.. lets see if it still works at the end of the day :)

theyosh, 2019-07-09:
Could you then do another git pull and restart? Made some small changes to log this error.

swekley, 2019-07-12:
no more errors, sensor works great now. thank you!

closed 2019-07-12 by swekley