theyosh/TerrariumPI #312: how to conect an analog Ph device?

funware84, 2019-07-09:
Hi,
i am very happy with this.
i am monitoring my pond for swimming with this.
it is without of chlorine

now i want to monitore pH-Value, but dont know what to order and how to connect.
may someone can post an introduction with some PIC's?

kind regards Florian

theyosh, 2019-07-09:
Could this help you: https://github.com/theyosh/TerrariumPI/issues/108

This needs a converter like: https://rasp.io/analogzero/

But I do not have this tested myself. So I am not sure if this will work together with other sensors due to space issues or occupied gpio pins

funware84, 2019-08-01:
HI,
thanks, i got it work with the skript but not in TerrariumPI, what have i to write into adress field?

theyosh, 2019-08-07:
Hmm, in Terrariumpi you should select the ph sensor, and use the analog port number. Your script should also use a analog port number. So that value needs to be in my software.

Another solution, is to use a script sensor, or let your script output some json and use that for remote sensor.

https://github.com/theyosh/TerrariumPI/wiki/Hardware#analog-sensors
https://github.com/theyosh/TerrariumPI/wiki/Remote-data

theyosh, 2019-09-10:
Any updates on this?

theyosh, 2019-10-01:
Closing due to no response

closed 2019-10-01 by theyosh