theyosh/TerrariumPI #314: HTU21D problem

swekley, 2019-07-28:
My HTU21D gives an input/output error, while it still works outside of terrariumpi..

`
2019-07-28 03:02:02,974 - WARNING - terrariumI2CSensor   - Error getting second part of data in bytes from sensor 'Htu' with address 40 with error: [Errno 121] Remote I/O error`

`pi@raspberrypi:~ $ ./HTU21D_test 
24.27C
76.66%rh`

swekley, 2019-08-05:
My MLX90614 is also bugging, the ambient temperature is stuck while infrared temperature still works

theyosh, 2019-08-07:
Hmm, this is strange. The only thing I can think off, is that my software does not close or terminate the connection. Because when you script is ended, all the connections are closed.

So could be that is the case. But that is a bit hard to test.

On the other hand, the code for MLX90614 is not written by me, so I can't tell why it is causing issues. And I do see in my code, that I do close the connections. So then I do not have any clue why this is happening...

Also, this is hardware, which also for me is hard to debug.... Not that known with all that small electronics.

So, try to reboot the Pi? Update software? If possible, just reconnect the sensors, so all connections are clean and solid....

theyosh, 2019-09-10:
Closing due to no reponse

closed 2019-09-10 by theyosh