theyosh/TerrariumPI #316: Syntax error while updating to 3.9.7

prcdslnc13, 2019-07-28:
I started the update and things were running fine until my terminal went red and froze. I exited out and restarted the ssh session and tried to re-run the update and now get this every time. I even tried a fresh git-clone

Current default time zone: 'America/Chicago'
Local time is now:      Sun Jul 28 11:25:27 CDT 2019.
Universal Time is now:  Sun Jul 28 16:25:27 UTC 2019.

./install.sh: 225: ./install.sh: Syntax error: "(" unexpected (expecting ")")

theyosh, 2019-07-28:
Hmm looks like a corrupted file.... here it is working fine. Try:

```
rm install.sh
git checkout install.sh
```
And retry to run the installer.

theyosh, 2019-07-28:
Hmmm, if a fresh clone is not working.... Then I am afraid that the SD card is corrupt. The installer is working for me on python 2 and 3.

prcdslnc13, 2019-07-28:
yea, the checkout didn't work either. Is there an easy way to port all my settings and configuration to a fresh Rasbian install?

theyosh, 2019-07-28:
Yesh! https://github.com/theyosh/TerrariumPI/wiki/FAQ#which-files-to-backup

Basically, backup those files, and put them in the new install and should be good to go

prcdslnc13, 2019-07-28:
Bah How did I miss that. Thanks so much man.

closed 2019-08-07 by theyosh