theyosh/TerrariumPI #317: Installation using Buster

dfester007, 2019-08-06:
Just wanted to share that I love the tool, however, the install was definitely a challenge until I STOPPED trying to use Raspian Buster-Lite and switched to Stretch-Lite.  This was completely my fault, as most of the past successful installs clearly were done by others using Stretch.  However, I just wanted to caution and share as Buster is the most up to date version and of course is what will install when users start with a tool like NOOBS.  The installation tools behavior with Buster was very different than with Stretch, it didn't actually perform a Python compile during the Buster install attempt.  Also, several modules continually failed to install on Buster, PYALSAAUDIO, LED.OLED, EMAILS.  Even manual installs would fail on Buster.  There were a couple of module installation failures on Stretch but manually installing did work... specifically EMAILS and ICALENDAR.

No need to keep this open, but I wanted to encourage fellow users to stick with it and if it helps, try starting with Strech not Buster.

theyosh, 2019-08-07:
Hmm, strange. Because I am developing on Buster :P So I will have to look into this.

I did notice that Buster has some different packages... But I thought that I had tackled that in the installer script.

You did an install with the latest code?

dfester007, 2019-08-08:
I can't be 100%.  But yes, I followed the install instructions, cloned the directory... ran updates... ran install... etc... I did the exact same with stretch and got totally different behavior.  All of the same hardware.  I wondered too if some of the module installation inconsistency has to do with US vs EU servers... shouldn't matter... but wondered.

speedracer1190, 2019-08-19:
Sure thing, Im actually doing an install now and started on saturday and was working my way through the missing modules.  
@theyosh did you commit to the masterbranch? ill download a freshy and give it a try

speedracer1190, 2019-08-19:
@theyosh I can confirm it works!  thanks heaps for the update.  It did take 2hrs 45mins on a raspberry pi B+ v2.

theyosh, 2019-08-19:
Cool! And sorry for the long installation.... :)

closed 2019-08-19 by theyosh

speedracer1190, 2019-08-19:
No need to be sorry.  Its great that its working. Thank you.