theyosh/TerrariumPI #318: Denkovi v1

dfester007, 2019-08-08:
I see that Denkovi V2 is now supported, I didn't know that at the time so I ordered V1.  I don't see V1 in the list when I add switched... is V1 no longer supported?  Am I missing something?  I read in the help file that it's supposed to be automatically detected when adding a switch?  No such behavior that I can see.  Suggestions?

dfester007, 2019-08-09:
I did figure this out.
For Denkovi V1 USB Relays, when you add a switch under settings, choose FTDI as the hardware type.  For the address, use an integer that matches the relay channel you are adding like on a 4 channel it would be 1, 2, 3, or 4.  Complete the rest of the parameters as you desire.  You do need to add each channel of the Relay if it is a multi Relay device.

closed 2019-08-09 by dfester007

theyosh, 2019-08-14:
Nice you found it... https://github.com/theyosh/TerrariumPI/wiki/Hardware#denkovi is the numbers documented, but I see it is missing that you have to select FTDI. Because that is the general way of communication for more devices. I will update the wiki