theyosh/TerrariumPI #319: Float valves

Geek16432, 2019-08-09:
Hello,

I've looked a few places but can't find which float valves are supported. I want to make sure my water reservoir for the mister doesn't run dry.

theyosh, 2019-08-09:
That is correct. I do not support those.

But most of us use the https://github.com/theyosh/TerrariumPI/wiki/Hardware#hc-sr04 sensor to measure a distance. And when used in the water tank environment, it will calculate to available litre water.

So we do it based on a distance.

Geek16432, 2019-08-09:
Okay. Thank you

theyosh, 2019-08-14:
But, do you have an example of what you want to use? As I know float valves, they need to be set open en closed by hand or electronics.

So, then it should be possible to connect valve to a power unit to toggle it open when powered, and closed when power down. And with a hc-sr04 sensor, you can detect the height between the sensor and top of the water level. When to high (so low water level) you can power the valve for X seconds so it will fill up.

Is this what you are looking for? Then you need to find a way to open en close a valve with a power switch.

Geek16432, 2019-08-14:
Actually, it doesn't need to power a valve. I want a simple notification when the water level is low.

I was planning to use the M16 variation shown here

https://www.amazon.com/Maxmoral-Plastic-Vertical-Aquariums-Measuring/dp/B07QRGTVYD/ref=sr_1_15?keywords=vertical%2Bfloat%2Bswitch&qid=1565811932&s=gateway&sr=8-15&th=1

theyosh, 2019-08-18:
Ah, yes, those should be able to support. But they do not provide useful information to my opinion. You can hook them up as a door sensor at the moment. And that should then just trigger open and close messages. https://www.cl.cam.ac.uk/projects/raspberrypi/tutorials/robot/buttons_and_switches/

But for a sensor, it will only tell full or not full. But not how much. So in order to use those, you should use two combined. So when both floats are reporting empty, add water. When the highest float will message that it is full, stop the water filling.

But that involves two sensors working together, in a special case. That is a bit hard to make. So using a height sensor, works easier. So, at the moment, I am thinking not to support those, and advise to use a HC-SR04

Geek16432, 2019-08-18:
I trust your opinion

closed 2019-08-18 by theyosh