theyosh/TerrariumPI #320: 2 Different heat sources using sensor

dfester007, 2019-08-09:
I have 2 different heat sources in my setup
1. Daytime - Halogen bulb
2. Nighttime - Ceramic heater

I can see how to control either ONE of these sources but not both.  Ideal setup is during daytime, I control the Halogen bulb(s) with a dimmer using say temp sensor #1 (low limit +95).  Then, at night, I control the ceramic heater with a second dimmer also using sensor #1 (low limit +80).

Would it work if I had on/off relays running on basic timers, establishing the day / night schedule, and then 2 dimmers in series with the on/off being driven by PWM all the time, but obviously  only the dimmer receiving power would effect the environment?  Would that work?  Is there a more elegant way to do this?  Anyway to add another temperature environment?   Or trick one of the unused environments to look at a temp sensor...?

Thanks.
Dennis

dfester007, 2019-08-14:
So I was able to work this scenario out.

1.  Hardware, I hooked up 2 identical circuits consisting of a relay and a dimmer control, one for my daytime bulbs and the other for my nighttime ceramic heater.
2.  In environment, configure the 'Lights'.  In daytime, turn on the daytime relay.  At night, turn on the night time relay.  This sets up the basic on/off times for the lighting.
3.  In environment, configure the 'temperature' for sensor mode.  The trick here is under the alarm settings to select BOTH of the dimmer switches for the power switch setting.  That way, regardless of time of day, the dimmers are being driven.

Seems to work just fine.  Letting it run a could of days just to be sure.  But thought I would share the power of what is possible with this setup.

closed 2019-08-14 by dfester007

theyosh, 2019-08-14:
Hmm, sorry for not responding earlier. Nice you found a solution.

I would suggest almost the same. Use the halogeen as a normal day light. This will heat up. And only use a dimmer for the ceramic heater during night time. But then, you cannot control the temperature during the day. Or, you have to set the max dimmer value on the switch itself, so it will keep a certain temperature.

I do know, that during the day I have enough heat from the light system. During the night, I use temperature sensors to keep it at a minimal temperature. So I only need dimming for the heating during the night.

But I do agree, it would be better if it would be possible to control 24/7 based on sensors, but with different hardware. At the moment, it is not possible right out the both.

So let me know how it turns out.

Natrixz, 2019-08-14:
Don't forget that the temperature can be different based on outside enclosure ambient temp (for example if its 35 degrees for days or if the sun hits the glass, etc)