theyosh/TerrariumPI #321: Installed master updated and a camera getting error now

dfester007, 2019-08-17:
Getting this segmentation fault on start.  Any suggestion?

./start.sh: line 125:  1419 Segmentation fault      python "${BASEDIR}/terrariumPI.py"
2019-08-17 18:49:45,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
2019-08-17 18:49:45,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 1 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2019-08-17 18:49:55,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://10.0.0.40:8090 ...
./start.sh: line 125:  1467 Segmentation fault      python "${BASEDIR}/terrariumPI.py"
2019-08-17 18:49:55,000 - INFO    - terrariumWrapper     - Restart counter: 2/5
2019-08-17 18:49:55,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 0 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2019-08-17 18:50:05,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://10.0.0.40:8090 ...
./start.sh: line 125:  1509 Segmentation fault      python "${BASEDIR}/terrariumPI.py"
2019-08-17 18:50:06,000 - INFO    - terrariumWrapper     - Restart counter: 3/5
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l

theyosh, 2019-08-18:
IF a reboot does not fix it, you have to re-install the SD card. I have no other solutions for this.

Keep a copy of the important files: https://github.com/theyosh/TerrariumPI/wiki/FAQ#which-files-to-backup

dfester007, 2019-08-26:
No reason to keep this open.  Rebuilding SD card

closed 2019-08-26 by dfester007