theyosh/TerrariumPI #322: Energenie PG-PMS 2 Error

themanwithoutplan, 2019-08-18:
Hi Theyosh,

After connection the Energenie PG-PMS2 with USB to the Raspbeery and creating a switch wih Energenie USB an Adress Value "1" the switch is not responding for manuel toggle or Timer. 

The Logfiles are below:

2019-08-18 15:37:42,004 - ERROR   - terrariumSwitch      - Failed changing power switch 'light 1' of type 'eg-pm-usb' at address '1' from state 'None' to state 'True' (Forced:False)
2019-08-18 15:37:39,719 - ERROR   - terrariumSwitch      - Failed changing power switch 'light 1' of type 'eg-pm-usb' at address '1' from state 'None' to state 'True' (Forced:False)
2019-08-18 15:37:39,305 - INFO    - terrariumEngine      - Update done in 0.96149 seconds. Waiting for 29.03851 seconds for next update
2019-08-18 15:37:39,175 - INFO    - terrariumSwitch      - Updated eg-pm-usb power switch 'light 1' status in 0.00010 seconds

Rebooting or complete new instalation doesn't help. 

Am i missing something out?

theyosh, 2019-08-18:
I can't tell. For it is working. So try to do a `git pull` and rerun the installer. Just to make sure everything is up to date.

And maybe reboot. If that does not help, check the connections? Could the usb cable not be connected fully?

themanwithoutplan, 2019-08-19:
Done a git pull und reinstalled/rebooted. Still not working. 

The switch is connected and reconized by the pi:

pi@raspberrypi:~ $ pi@raspberrypi:~ $  lsusb
Bus 001 Device 020: ID 04b4:fd15 Cypress Semiconductor Corp.
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

theyosh, 2019-08-19:
Ok, this is strange. You have a sligtly newer version then I have. But that should not be an issue. So could you check the following (assuming you use the user pi):

`cat /etc/group | grep pi` and past the outcome.

`cat /etc/udev/rules.d/60-sispmctl.rules` and past the outcome.

run as normal user: `/usr/bin/sispmctl -d 0 -o 1` and what is the outcome of this.

themanwithoutplan, 2019-08-19:
Ok, here we go: 

cat /etc/group | grep pi:

adm:x:4:pi
dialout:x:20:pi
cdrom:x:24:pi
sudo:x:27:pi
audio:x:29:pi
video:x:44:pi
plugdev:x:46:pi
games:x:60:pi
users:x:100:pi
input:x:101:pi
netdev:x:108:pi
pi:x:1000:
spi:x:999:pi
i2c:x:998:pi
gpio:x:997:pi
sispmctl:x:1001:pi

cat /etc/udev/rules.d/60-sispmctl.rules:

SUBSYSTEM=="usb", ATTR{idVendor}=="04b4", ATTR{idProduct}=="fd10", GROUP="sispmctl", MODE="660"
SUBSYSTEM=="usb", ATTR{idVendor}=="04b4", ATTR{idProduct}=="fd11", GROUP="sispmctl", MODE="660"
SUBSYSTEM=="usb", ATTR{idVendor}=="04b4", ATTR{idProduct}=="fd12", GROUP="sispmctl", MODE="660"
SUBSYSTEM=="usb", ATTR{idVendor}=="04b4", ATTR{idProduct}=="fd13", GROUP="sispmctl", MODE="660"
SUBSYSTEM=="usb", ATTR{idVendor}=="04b4", ATTR{idProduct}=="fd15", GROUP="sispmctl", MODE="660"

/usr/bin/sispmctl -d 0 -o 1:

No GEMBIRD SiS-PM found. Check USB connections, please!

theyosh, 2019-08-19:
Hmm, the last command failed. What happens if you run it with sudo? Does it then find the GEMBIRD device?

theyosh, 2019-08-19:
Found it.... apparently the sispmctl package is still outdated... see here: https://github.com/theyosh/TerrariumPI/issues/195

So, you have to install the sispmctl package from source to support your USB device

themanwithoutplan, 2019-08-19:
Hm, strange, updated everything a few times. 
Used the workaround from #195 works fine now. 

Thank you a lot.

theyosh, 2019-08-19:
Ok, then I'll close this issue. A bit of a shame that the packages are still not up to date... :(

closed 2019-08-19 by theyosh