theyosh/TerrariumPI #323: Fresh Installation all modules appear present... get this error

dfester007, 2019-08-22:


dfester007, 2019-08-22:
pi@raspberrypi:~ $ screen -r
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumCalendar.py", line 8, in <module>
    from icalevents.icalevents import events
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/icalevents/icalevents.py", line 3, in <module>
    from .icalparser import parse_events
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/icalevents/icalparser.py", line 130
    def encode(value: Optional[vText]) -> Optional[str]:
                    ^
SyntaxError: invalid syntax
2019-08-21 23:19:34,000 - INFO    - terrariumWrapper     - Restart counter: 4/5
2019-08-21 23:19:34,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 5 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!

dfester007, 2019-08-22:
2019-08-21 23:22:38,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://10.0.0.40:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 23, in <module>
    from terrariumEngine import terrariumEngine
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 38, in <module>
    from terrariumCalendar import terrariumCalendar
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumCalendar.py", line 8, in <module>
    from icalevents.icalevents import events
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/icalevents/icalevents.py", line 3, in <module>
    from .icalparser import parse_events
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/icalevents/icalparser.py", line 130
    def encode(value: Optional[vText]) -> Optional[str]:
                    ^
SyntaxError: invalid syntax
2019-08-21 23:22:43,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
2019-08-21 23:22:43,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 5 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7

dfester007, 2019-08-22:
Not very good at debug on Python, but I did notice that icalevents package might be the source of the error and that the module was updated just yesterday... someone with more experience / skill might see something.

dfester007, 2019-08-26:
Well, just for kicks I tried a completely fresh install, different SD and PI... got to the same place... I can only assume it's the icalparser.py module...  trying to downgrade the version now.

2019-08-26 08:01:03,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://10.0.0.6:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 23, in <module>
    from terrariumEngine import terrariumEngine
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 38, in <module>
    from terrariumCalendar import terrariumCalendar
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumCalendar.py", line 8, in <module>
    from icalevents.icalevents import events
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/icalevents/icalevents.py", line 3, in <module>
    from .icalparser import parse_events
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/icalevents/icalparser.py", line 130
    def encode(value: Optional[vText]) -> Optional[str]:
                    ^
SyntaxError: invalid syntax
2019-08-26 08:01:08,000 - INFO    - terrariumWrapper     - Restart counter: 4/5
2019-08-26 08:01:08,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 5 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!

MP98n, 2019-08-26:
I’m having exactly the same issue.  I’ve tried two SD cards and several fresh installs with no success

dfester007, 2019-08-26:
I had a little success this AM before heading into work.

I uninstalled icalevent version 0.1.22, and specifically installed 0.1.21 that seemed to move things forward.  Of course I rebooted the pi after the uninstall of 0.1.22 and after the install of 0.1.21.

I just think there were some changes to the module that Yosh couldn't have been aware of... or the module is just busted.

theyosh, 2019-08-26:
Yeah, I fond this out also yesterday. Did not had time to fix the installer.

The icalevents python module is official only python3. But it would install until version 0.1.21 . So I have pinned that version for Python2. Python3 should just work with the latest version.

Do a `git pull` and rerun the installer. It should work now.