theyosh/TerrariumPI #324: reboot error will not start

speedracer1190, 2019-08-24:
Hi there,
Firstly i want to say thank you for your work,  this is an amazing project.  I have been having issues trying to get the DHT22 or DHT11 to work. 
But I just went to restart it and I am getting this error in the screen -r
 File "/home/pi/TerrariumPI/terrariumSensor.py", line 88, in get_sensor_cache_key
    self.__sensor_cache_key = md5((self.get_type() + self.get_address()).encode()).hexdigest()
TypeError: cannot concatenate 'str' and 'NoneType' objects
I have a dallas 1wire temp sensor which was working connected, is it something to do with that?

funware84, 2019-08-24:
I had same problem with an broken 1wire temp sensor.
As it is in settings.cfg and the system can not find the hardwarw while booting. It will result in an error.
I did `sudo` nano settimgs.cfg `and` deletet all lines which belong to this 1wire sensor

speedracer1190, 2019-08-25:
Thanks @funware84 ill give it a shot.  was simply removing it and rebooting not enough?

theyosh, 2019-08-27:
@speedracer1190 what kind of issues. DHTXX sensors are not stable: https://github.com/theyosh/TerrariumPI/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aclosed+dht

theyosh, 2019-08-27:
The code has been fixed for the startup issue. So a missing 1wire will not stop the software from starting up. About the DHT sensors, look at the closed issues... you will see a lot of problems due to hardware. So for humidity, there are different options. Look at the Wiki for supported humidity hardware.

theyosh, 2019-09-10:
Any updates on this?

theyosh, 2019-10-01:
Closing due to no response

closed 2019-10-01 by theyosh