theyosh/TerrariumPI #325: Use systemctl for startup scripts?

victorhooi, 2019-08-25:
What are the thoughts on integrating with systemd for startup/shutdown/reboot scripts etc?

Its the default system/service manager in Debian since Jessie.

That way, we can use systemctl for managing startup/shutdown/reoboot, and use journalctl to view/manage the process logs.

theyosh, 2019-09-10:
Hmm, I have made some first attempts, but my software does not play nice with systemd :( So I have to make some changes.

Bigest problem is that I currently depends on a temp folder which created when the user logged in. But with systemd there is no such login, and I do not have that temporary folder. Without it, I have to make all logging to the SD card and that will wear it down....

So I have to look into it a bit more.

theyosh, 2019-11-10:
Hi,

I added https://github.com/theyosh/TerrariumPI/blob/master/contrib/terrariumpi.service

to my software. With that, I am able to run my software as a service.

I installed by making a symlink:
`ln -s /home/pi/TerrariumPI/contrib/terrariumpi.service /lib/systemd/system/terrariumpi.service`

`systemctl daemon-reload`

And then you should be able to enable it by:

`systemctl enable terrariumpi`

vacuumboots, 2019-12-17:
This seems like a good service theyosh, just want to say i using the terrariumpi.service as you have is working well for me on Raspbian lite Buster and Python 3

theyosh, 2020-01-28:
Ok, I am closing this, as I have added information in the Wiki how to use this. Maybe in later version this will be the default. https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-systemd

Thanks for testing.

closed 2020-01-28 by theyosh