theyosh/TerrariumPI #326: Post latest update.  TerrariumPI crashes

jkissner, 2019-08-28:
^
SyntaxError: invalid syntax
2019-08-28 16:28:25,000 - INFO    - terrariumWrapper     - Restart counter: 4/5
2019-08-28 16:28:25,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 7 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3 2 1 restart!
2019-08-28 16:28:35,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://10.251.5.191:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 23, in <module>
    from terrariumEngine import terrariumEngine
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 38, in <module>
    from terrariumCalendar import terrariumCalendar
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumCalendar.py", line 8, in <module>
    from icalevents.icalevents import events
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/icalevents/icalevents.py", line 3, in <module>
    from .icalparser import parse_events
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/icalevents/icalparser.py", line 130
    def encode(value: Optional[vText]) -> Optional[str]:

theyosh, 2019-08-30:
This is happening with the latest install.sh script? I am not sure if pip will downgrade when you add a version number. So it could be that a manual action is needed. In that case do the following:

`sudo pip remove icalevents`
`sudo pip install icalevents==0.1.21`

And restart the terrariumpi software.

theyosh, 2019-09-10:
Any updates on this? Did above fix your issue?

theyosh, 2019-10-01:
Closing due to no response

closed 2019-10-01 by theyosh

jkissner, 2019-10-02:
It seems to have fixed the problem.  Thank you!