theyosh/TerrariumPI #329: Motion detection without archive?

Swiftnesses, 2019-09-15:
Howdy,

Is it possible to only enable motion detection notifications, without having the archive any images? I'd love to know when my ducks enter the coop at night - but don't need archived images (the most recent would be good, but not essential).

theyosh, 2019-10-01:
Hi,

the funny ducks again :P. Still makes me smile : https://user-images.githubusercontent.com/1001877/49548915-23fd5300-f8df-11e8-8f36-68a820b171b9.png

The answer is no and yes :D. It is a bit hard to make this in code, so the easy trick here is to use a cronjob that just deletes the archived images.

I added a cronjob example in the contrib folder. Add in the the cron.daily folder and it should clear it up.

Swiftnesses, 2019-10-01:
🤣🤣🤣🤣

They're wicked pets! At least they brought a smile to your face 👍 

I'll give this a shot and reopen if I run into issues, thank you for such an awesome tool.

closed 2019-10-01 by Swiftnesses