theyosh/TerrariumPI #330: Conductivity

rrgrow, 2019-09-20:
Hi Again
I have just added an EC meter to my arduino which sends through usb serial to a json file on pi.
The only problem is when adding the sensor it will not allow me to use "remote" and "Conductivity" together, it doesn't save. So I tried "Remote" & "Fertility" and it works. 
Not sure if this is a bug or you intend to do this.
 
 Also would it be easy enough to add a "Calibration" field to the sensors section +- values as some sensors can be out slightly.
Many Thanks

theyosh, 2019-09-24:
Hmm, it should be supported through remote. So I have to look into that.

Calibration is something I have done with the Chirp sensor. So I can re-use for other sensors. Have to look into it.

theyosh, 2019-10-01:
Hmm, that is strange. I has never worked since the last rewrite :)

So I added just 1 line to support remote conductivity... :P So do a git pull and restart. It should work now. Thanks for noticing it.

theyosh, 2019-11-10:
Any updates? Is it working for you?

theyosh, 2019-12-02:
Closing due to no more response

closed 2019-12-02 by theyosh