theyosh/TerrariumPI #331: I2c

scrumpydc, 2019-09-23:
Can it be used with i2c products like pca9685 for controlling led lights using ldd driver  and a bme280 temp and humidity sensor this would be an amazing feature not so much an issue but would be cool if you got this working with own controll for day and night cycle

theyosh, 2019-09-24:
Hi,

the bme280 sensor is already supported. So that one can be used.

the pca9685 are 16 dimmers!! :) That is funky. 
@AndrejWeber : What are your thoughts on this?

It looks like it can be supported. But I cannot test it. So if you have the hardware, I can make a test branch to test it.

https://github.com/voidpp/PCA9685-driver

AndrejWeber, 2019-09-24:
Hi,
The pca9685 should work with same logic like the DC-Dimmer, but you do not have to set up a PWM signal on a GPIO, but sent some i2c commands to set up the outputs on this board. 

The bigger problem is, that this board can only drive a single LED for each output. To use a 12V - LED-stripe you need a additional led-driver, that supports PWM dimming.
So it depends, how much independent lights you want to use. 
If they should dim all together, you can also use the DC-Dimmer to control the LED driver.
Can be, I do not understand what you wish to do, but I do not see a reason, to set up 16 different light values at same time.

scrumpydc, 2019-09-25:
Well I just got it running with node red you use this to controll ldd drivers witch inreturn controlls high power LEDs there is a product called stormx controller witch is basically a nano  with a pca9685 I have it set on node red to fade in and out with day and night then when the mistaking comes on it flashes to emulate a storm but that is as far as node red can take me all the pca9685 needs is a number between 0 -4095

theyosh, 2020-01-28:
Ok, I added support for PCA9685. Do a `git pull` and restart. Make sure you clear your browser cache.

Then you can add the PCA9685 dimmer at the power switches settings page.

Add an address in the like: [I2C bus number],[LED Nr],[LED Nr],[LED Nr]

Where [I2C bus number] = 40 and [LED Nr] 0 - 15 https://github.com/voidpp/PCA9685-driver

You can add multiple leds per dimmer, by adding multiple [LED Nr] in the address

I am not sure what will happen when you add this dimmer twice with different led nrs. I am not sure if the PCA device can be shared in code.... can't test it.

theyosh, 2020-03-14:
Did you have any time yet to test this? Any updates?

theyosh, 2020-03-26:
Final reminder... any updates?

scrumpydc, 2020-03-26:
No sorry I have not had a spare raspberry pi

theyosh, 2020-03-26:
No problem. But then I close it for now. If you have time to test it, and it does not work, you can reopen this. Or create a new issue.

closed 2020-03-26 by theyosh

scrumpydc, 2020-03-26:
Ok np thank you