theyosh/TerrariumPI #332: Fails to get SHT31 sensor values

Died115, 2019-09-25:
Hello,

Upgraded my terrarium with the SHT31 sensor but saw in the logs that the sensor returns no value.
Used address '44' 
Checked with i2c-detect -y 1 and the sensor responds on 0x44
Checked with a different python test script, the sensor values get pulled correctly.
I've seen that you've been using the older Adafruit_SHT31 library which uses 8 bit data chunks. Whilst the sensor uses 16 bit datachunks. That way the heater cannot be read out properly.

[image]

Kind Regards,

theyosh, 2019-10-01:
Hi,

Could you upload your script here? So I can compare and update my code.

theyosh, 2019-11-10:
As I cannot replicate this, and you did not provide your used testing script, I cannot fix this. Could you please upload your testing script?

Died115, 2019-11-10:
Hello,
I was on holiday so my apologies if wasn't able to reply sooner.
The testing script I used originated from github:

https://github.com/ralf1070/Adafruit_Python_SHT31

The explanation to this error i found on stack overflow:
https://stackoverflow.com/questions/35461155/adafruit-sht31-d-and-raspberry-pi2-unable-to-read-data-from-sensor

Hopefully this gives you some more info :) 

Thanks for the great work on TerrariumPi :)

theyosh, 2019-11-11:
Hi,

this is unclear to me. I use the library you are referring to: https://github.com/ralf1070/Adafruit_Python_SHT31 . So that is installed

Then my question is, what is the new library where it is using 16bit data blocks? As the library on github does not do that as far I can see.

Are you saying that the script: https://github.com/ralf1070/Adafruit_Python_SHT31/blob/master/Adafruit_SHT31_Example.py is working for you? Because that code is the same as I use. So then I am afraid that something else is causing issues.

Or did you use a different script to test? It is unclear how you got it working. As I am doing the same as described on the github page. I do not own this kind of hardware, so I am not able to test it.

Died115, 2019-11-12:
Hello,

The example gave the heater error as well. But was able to get some temperature data.
There was one script that was able to acces the internal heater and get complete readings from the sensor.

https://github.com/adafruit/Adafruit_CircuitPython_SHT31D

Here is the script that worked to the full extend.

Kind Regards,

theyosh, 2019-11-18:
Hmm, it looks like a different sensor... so the question is then, will it also support older sensors? I do not own  one, so I can't test.

I think I will just add it, as a new sensor. Just to make sure it will not break existing setups

theyosh, 2019-12-02:
Ok, I added support for SHT31D sensor. Please do a `git pull` and run the installer. Make sure you have the system running on Python3. As this library is only supported for Python3.

`sudo ./install.sh 3`

Died115, 2020-01-20:
Doing a new terrarium build in February using a SHT31D. 
Then i will give you an update. 
I will not modify existing setups because then i have to move the animal to a different (less equipped) enclosure, to do the reconfiguration.

Kind Regards,

Died115, 2020-02-07:
[image]

I get this when i try to install with the <3> parameter to use python3

theyosh, 2020-02-07:
Hmm, strange. If you have the latest code and rerun the installer, that missing module should be installed. 

But to do that manually run the following command: `sudo pip3 install PCA9685-driver`

And restart the software. That should fix the missing module problem.

jeff26300, 2020-02-08:
Hello,

I have just installed a SHT31D sensor as well but I cannot read out neither the temperature and the humidity through TerrariumPi, while I can get the values via the command. The sensor is on address 44 and I executed the command above (sudo pip3 install PCA9685-driver). After rebooting the software, the values are still not read :(

[image]

Thanks!

theyosh, 2020-02-08:
@jeff26300 Could you provide some debug: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

Else I can't see what is going wrong now.

jeff26300, 2020-02-09:
Thanks @theyosh , I will look into it, I need some time to learn how to use these commands, I am a really beginner in Raspberry world, I keep you posted

jeff26300, 2020-02-09:
So, it finally went faster than I expected! Below is the screen I got
It mentions the DHT22 sensor, that's weird. 

[image].

jeff26300, 2020-02-09:
So, it finally went faster than expected! I managed to get the screen below with errors about the sensor
[image]

I checked again the i2c address and it is 44, as per the screen shot below:

[image]

44 is the value I put in the address field of the sensor SHT3XD in TerrariumPI.

theyosh, 2020-02-09:
Hm, I am afraid you are running it with Python 2 (default) which does not support that sensor.

You should install it with Python3: `sudo ./install.sh 3` and then start it with 'sudo ./start.sh 3'

And change the startup according to: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35

jeff26300, 2020-02-09:
That's great @theyosh !! It now works beautifully!! Thanks again for your great support !

theyosh, 2020-02-09:
No problem! And you can just stick to Python3 setup. That will/should work the same as the Python2 version. So have fun with it!

theyosh, 2020-03-14:
Still working? So we can close this issue?

jeff26300, 2020-03-15:
For me yes, perfectly! However I don't know if this is solved for @Died115

Died115, 2020-03-15:
Yes for me it is now completely operational

theyosh, 2020-03-15:
Thanks for the feedback. Closing now.

jeff26300, 2021-10-30:
Hello, I run the procedure and I get the following:

2021-10-30 17:38:23,186 - ERROR   - hardware.sensor       - Error updating sensor HC-SR04 ultrasonic ranging sensor distance named 'Water level' at address '23,24'. Check your hardware! Invalid reading from sensor HC-SR04 ultrasonic ranging sensor distance named 'Water level' at address '23,24'
2021-10-30 17:38:23,187 - WARNING - terrariumEngine       - Could not take a new measurement from sensor hc-sr04 distance named 'Water level' at address '23,24'. Tried for 16.05 seconds. Skipping this update.
Open sht3x I2C bus
sht3x start write action
Done writing, wait 0.2 sec
sht3x start read action
Done reading, data:
[100, 35, 130, 249, 143, 14]
2021-10-30 17:38:23,393 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor humidity named 'Humidity' at address '44' has exception: 0
2021-10-30 17:38:23,394 - WARNING - hardware.sensor       - Invalid reading from sensor Sensirion SHT3X sensor humidity named 'Humidity' at address '44', retrying in 0.5 seconds...
Open sht3x I2C bus
sht3x start write action
Done writing, wait 0.2 sec
sht3x start read action
Done reading, data:
[100, 24, 173, 249, 149, 150]
2021-10-30 17:38:24,099 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor humidity named 'Humidity' at address '44' has exception: 0
2021-10-30 17:38:24,100 - WARNING - hardware.sensor       - Invalid reading from sensor Sensirion SHT3X sensor humidity named 'Humidity' at address '44', retrying in 0.5 seconds...
Open sht3x I2C bus
sht3x start write action
Done writing, wait 0.2 sec
sht3x start read action

theyosh, 2021-10-30:
Damm, that was a stupid bug... So do a `git pull` and restart. I think it should work...

jeff26300, 2021-10-30:
It now works beautifully!! Thanks a lot

Jeff

theyosh, 2021-10-30:
Nice!. Closing it again :)

theyosh, 2022-04-13:
If you did a `git pull` you have to rerun the installer `sudo ./install.sh`. Because you are now missing a Python library. 

So I get the feeling that you did a git pull but not a rerun of the installer.

But, to be sure, I advise now to do a `git pull` to make sure you are on the latest version (4.3.0) and run the installer `sudo ./install.sh` and then you can just run it in debug mode to see if it works. Then you can start the service again

jeff26300, 2022-04-14:
Ok! I redid what you said and obtain the following error messages. Any idea?

(venv) pi@Hegter:~/TerrariumPI $ python terrariumPI.py
2022-04-14 20:31:46,999 - INFO    - terrariumEngine       - Starting TerrariumPI 4.3.0 ...
2022-04-14 20:31:47,060 - INFO    - terrariumEngine       - Loaded 28 settings in 0.06 seconds.
2022-04-14 20:31:47,525 - INFO    - terrariumEngine       - Loading existing sensors from database.
2022-04-14 20:31:47,553 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named 'Température' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
2022-04-14 20:31:47,557 - WARNING - hardware.sensor       - Invalid reading from sensor Sensirion SHT3X sensor temperature named 'Température' at address '44', retrying in 0.5 seconds...
2022-04-14 20:31:48,081 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named 'Température' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
2022-04-14 20:31:48,081 - WARNING - hardware.sensor       - Invalid reading from sensor Sensirion SHT3X sensor temperature named 'Température' at address '44', retrying in 0.5 seconds...
2022-04-14 20:31:48,607 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named 'Température' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
2022-04-14 20:31:48,608 - ERROR   - hardware.sensor       - Error updating sensor Sensirion SHT3X sensor temperature named 'Température' at address '44'. Check your hardware! Invalid reading from sensor Sensirion SHT3X sensor temperature named 'Température' at address '44'
2022-04-14 20:31:48,610 - WARNING - terrariumEngine       - sht3x temperature named 'Température' at address '44' had problems reading a new value during startup in 1.08 seconds. Will be updated in the next round.
2022-04-14 20:31:48,612 - INFO    - terrariumEngine       - Scanning for new sensors ...
2022-04-14 20:31:57,798 - INFO    - terrariumEngine       - Loaded 1 sensors in 10.27 seconds.
2022-04-14 20:31:57,799 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-04-14 20:31:58,046 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Light' at address '11' value 0.00 in 0.23 seconds.
2022-04-14 20:31:58,053 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Heating light' at address '12' value 0.00 in 0.01 seconds.
2022-04-14 20:31:58,059 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Water pump' at address '13' value 0.00 in 0.01 seconds.
2022-04-14 20:31:58,463 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Fans' at address '15' value 100.00 in 0.40 seconds.
2022-04-14 20:31:58,465 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2022-04-14 20:32:09,610 - INFO    - terrariumEngine       - Loaded 4 relays in 11.81 seconds.
2022-04-14 20:32:09,611 - INFO    - terrariumEngine       - Loading existing buttons from database.
2022-04-14 20:32:09,620 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.01 seconds.
2022-04-14 20:32:09,621 - INFO    - terrariumEngine       - Loading existing webcams from database.
2022-04-14 20:32:09,628 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.01 seconds.
2022-04-14 20:32:09,629 - INFO    - terrariumEngine       - Loading existing enclosures from database.
2022-04-14 20:32:09,651 - INFO    - terrariumArea         - Updated area Lighting named Lights in timer mode at enclosure Hegter in 0.00 seconds.
2022-04-14 20:32:09,652 - INFO    - terrariumArea         - Refreshing timer table for Humidity named Rain period: high
2022-04-14 20:32:09,653 - INFO    - terrariumArea         - Updated area Humidity named Rain in timer mode at enclosure Hegter in 0.00 seconds.
2022-04-14 20:32:09,654 - INFO    - terrariumArea         - Updated area Moisture named Fans in timer mode at enclosure Hegter in 0.00 seconds.
2022-04-14 20:32:09,655 - INFO    - terrariumArea         - Updated area Heating named Heating Light in timer mode at enclosure Hegter in 0.00 seconds.
2022-04-14 20:32:09,655 - INFO    - terrariumEngine       - Loaded Enclosure Hegter with 4 areas in 0.02 seconds.
2022-04-14 20:32:09,656 - INFO    - terrariumEngine       - Loaded 1 enclosures in 0.03 seconds.
2022-04-14 20:32:09,736 - INFO    - terrariumEngine       - TerrariumPI is up and running at address: http://0.0.0.0:8090 in 24.72 seconds.
2022-04-14 20:32:09,000 - INFO    - terrariumWebserver    - Running webserver at 0.0.0.0:8090
2022-04-14 20:32:10,012 - WARNING - terrariumEngine       - Could not take a new measurement from sensor sht3x temperature named 'Température' at address '44'. Tried for 0.00 seconds. Skipping this update.
2022-04-14 20:32:40,046 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named 'Température' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
2022-04-14 20:32:40,048 - WARNING - hardware.sensor       - Invalid reading from sensor Sensirion SHT3X sensor temperature named 'Température' at address '44', retrying in 0.5 seconds...
2022-04-14 20:32:40,973 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named 'Température' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
[... cut ...]
2022-04-14 20:35:11,465 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named 'Température' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
2022-04-14 20:35:11,466 - ERROR   - hardware.sensor       - Error updating sensor Sensirion SHT3X sensor temperature named 'Température' at address '44'. Check your hardware! Invalid reading from sensor Sensirion SHT3X sensor temperature named 'Température' at address '44'
2022-04-14 20:35:11,468 - WARNING - terrariumEngine       - Could not take a new measurement from sensor sht3x temperature named 'Température' at address '44'. Tried for 1.44 seconds. Skipping this update.

theyosh, 2022-04-15:
Ok, it turns out that the SHT3XD does work different then the other SHTXX sensors. So I have added a new sensor called SHT3XD with the old code that was working in TP3.

So now, you should do a `git pull` once more, and rerun the installer `sudo ./install.sh` so that the new SHT3XD drivers are installed. Then you can start TP4 in debug. You need to delete the existing SHT3X sensor and then add a SHT3XD sensor. And see if that works better.

jeff26300, 2022-04-20:
It now works perfectly, both the temp and humidity are read! :) thanks again a lot for your support!
Have a nice evening

theyosh, 2022-04-20:
Nice!

commit eb9ba88a (2021-10-30): Clear debug. Fix #332
closed 2022-04-20 by theyosh