theyosh/TerrariumPI #333: livestream with usb webcam

SAM4191, 2019-09-30:
I am running TerrarirumPI on my RPi 2B. Installation was straightforward without any problems.

I've read these tutorials from stuart:
https://medium.com/@jinky32/using-the-raspberrypi-camera-with-terrariumpi-and-web-streaming-your-reptile-4c6676bbb1aa
https://medium.com/@jinky32/adding-a-camera-to-raspberrypi-for-use-in-terrariumpi-9629b027c746

He writes "It is possible to set up a streaming camera which uses HLS. In order to do this you go through exactly the same process as above but in the location field you need to use rpicam_live rather than /dev/Video0 in the location field.
Does rpicam_live mean it has to be an rpi-camera or should usb-webcams work as well?
So far I haven't got the live video working with my USB-Webcam (Microsoft livecam 3000).

When I change the Location to rpicam_live and restart TerrariumPi I get "Error code: hls:networkError_manifestLoadError".
I tried clearing the cache and using a different browser.

The log from screen -r looks fine to me.
terrariumWrapper     - Starting TerrariumPI server at location: http://:8090 ...
2019-09-24 22:57:10,685 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
2019-09-24 22:57:11,136 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
2019-09-24 22:57:19,000 - INFO    - terrariumWebserver   - Running webserver at :::8090

[image]

theyosh, 2019-10-01:
Hi,

unfortunately, live streaming is only available with the Raspberry PI camera. USB camera's can only be used as a static webcam.

Or you have to setup a second RPI that only does the transcoding of the USB webcam to HLS. But this is rather hard, and takes a lot of CPU power. So, I would stick to the static webcam for your USB webcam.

theyosh, 2019-10-01:
More info here: https://github.com/theyosh/TerrariumPI/issues/223#issuecomment-445011379

SAM4191, 2019-10-01:
Thanks for your answer
I'm gonna buy an rpicam and use the USB as static then.
Will add pics of my terrarium soon.

closed 2019-10-01 by SAM4191