theyosh/TerrariumPI #334: Webcam motion doesn't trigger notifications?

Swiftnesses, 2019-10-06:
I'm back...

After some testing it appears I cannot get notifications to fire based on motion.

I see the following in the logs:

`Saved webcam Coop Interior image for archive due to motion detection`

Sadly, notifications don't fire - I tried display and webhooks without success, any ideas?

Swiftnesses, 2019-10-06:
Update, tested via email and it works.

I assume this is because it sends an image and the webhook doesn't support that?

Swiftnesses, 2019-10-07:
@theyosh can you help me solve this, my ducks would appreciate their closed earlier 👍

theyosh, 2019-10-08:
Hmm, I think you are correct. The webhook does not have an option to post files... I will look into that, and will add something like a new variable called 'attachements' which is an array with file meta and base64 data... 

So, I think due to the complexity of sending binary data, it did not make it the first round. But we can have a second look and see how to we can fix this.

Swiftnesses, 2019-10-09:
Hey @theyosh that would be really awesome - currently I have no reliable way to tell when they come in at night, would be really useful.

Thank you as always :-)

Swiftnesses, 2019-10-11:
I tried placing a z-wave sensor out there for now, sadly it didn't last long before it was being sat on...

Lucky the Pi is mounted in a hard, waterproof case, hoping you can find time for a fix...

🦆 🦆 🦆

theyosh, 2019-10-13:
Hi, I made an update. Just do a `git pull` and restart. As I have not tested the code fully :( could you see if it

1. still works :)
2. does have a field names 'files' witch holds an array with files. Each file entry has two fields:
  - name = original file name from the pi
  - data = the image in base64 string.

Swiftnesses, 2019-10-13:
Did a `git pull` and now get:

```
2019-10-13 21:05:30,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.1.48:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 21, in <module>
    from terrariumEngine import terrariumEngine
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 30, in <module>
    from terrariumSensor import terrariumSensor
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 442, in <module>
    from terrariumBluetoothSensor import terrariumMiFloraSensor, terrariumMiTempSensor
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
  File "/home/pi/TerrariumPI/terrariumBluetoothSensor.py", line 8, in <module>
    from btlewrap.bluepy import BluepyBackend
  File "/usr/local/lib/python2.7/dist-packages/gevent/builtins.py", line 96, in __import__
    result = _import(*args, **kwargs)
ImportError: No module named btlewrap.bluepy
2019-10-13 21:05:36,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
```

Hmmm?

theyosh, 2019-10-13:
yes, my mistake. Another git pull and restart please

Swiftnesses, 2019-10-13:
Good man, it's started! Now let's see if those ducks want to move (they're sleeping now).

If not, I'll go out and hand test it tomorrow ;-)

theyosh, 2019-11-10:
Hi, any updates on this? Is this working for you?

Swiftnesses, 2019-11-16:
Hi Yosh,

Sorry for the late reply, doesn't appear to be sending the webhook here :-(

theyosh, 2019-12-02:
Hi, made an update... took some time. But I get the data now through webhooks.

You will have a new item called 'files' That should be an array with objects that contains a 'name' field which holds the original filename, and a 'data' field which holds the file in base64 encoded.

So you need to read the data part, decode it and save it.

Swiftnesses, 2019-12-02:
Testing now... I activated email and webhooks - I get the email, but no data to via the webhook?

theyosh, 2019-12-02:
Completely nothing? That sounds like a different problem.

You have entered the webhook url on top of the notification page?

Swiftnesses, 2019-12-03:
Yes, I have entered the webhook details (I get webhook data for other items already such as door open / close and switch change statuses).

theyosh, 2019-12-03:
Thanks, then I expect that there should be an error in the screen session. So could you have a look there. And see if there is some output?

theyosh, 2020-01-28:
Any updates....? (2x)

Swiftnesses, 2020-01-28:
Hey @theyosh... not yet, last time I checked I wasn't getting the motion notification via the webhook :(

theyosh, 2020-01-30:
Ok, after much investigating I found the following:

With Python3 (on Debian Stretch) the motion detection and archiving was broken due to outdated code in my software. So if you are running on Python3, I would expect that you have errors in the screen session.

With Python2 it just worked for me. I have added a PHP example file for parsing my Webhook data as a reference: https://github.com/theyosh/TerrariumPI/blob/master/contrib/receiving_webhook.php

So if you are running on Python3, please do a `git pull` and rerun the installer. Then restart TerrariumPI and everything should work.

If you are running with Debian Buster, there could be some issues... but at the moment I am not able to test :(

Swiftnesses, 2020-01-30:
I'll try this and let you know, thanks for the hard work (as always!).

Swiftnesses, 2020-01-30:
Oh, I just realised the installer states I have Python2 :-/

theyosh, 2020-01-30:
Hmm, I have updated the Wiki a bit. Maybe it is now clearer. But I do receive webcam image through motion detection. So I have to say that it works in Python2 and Python3 on Debian Stretch. At the moment I am not able to test on Buster. But for Python2 it should just work. Only with Python3 there could be an issue on Buster.

But do you not have any messages on the screen session?

Swiftnesses, 2020-01-30:
Pulled and reinstalled - lost my DHT sensors... will check tonight once home.

theyosh, 2020-01-30:
Crap... sorry man.... Lets hope it will comeback after a physical power cycle.

Swiftnesses, 2020-01-30:
Good idea, I switched the z-wave socket off and on and they returned! No emergency :-)

Ducks will be later, I'll see if the websocket is triggered by their motion!

Swiftnesses, 2020-01-30:
I think the motion is now working, NodeRed received a message with image info etc!

Swiftnesses, 2020-01-30:
`{"resolution_width":1296,"motiondeltathreshold":25,"image":"webcam/a94df035fc91a158c1f9d7a1c769ca66/a94df035fc91a158c1f9d7a1c769ca66_raw.jpg","max_zoom":3,"motionboxes":true,"id":"a94df035fc91a158c1f9d7a1c769ca66","archivelight":"ignore","archive":"motion","archivedoor":"open","last_update":"Thu Jan 30 16:13:02 2020","state":"online","location":"rpicam_live","preview":null,"archive_images":"","files":[{"data":"/9j/4AAQSkZJRgABAQAAAQABAAD/2wBDAAIBAQEBAQIBAQECAgICAgQDAgICAgUEBAMEBgUGBgYFBgYGBwkIBgcJBwYGCAsICQoKCgoKBggLDAsKDAkKCgr/2wBDAQICAgICAgUDAwUKBwYHCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgoKCgr/wAARCAPMBRADASIAAhEBAxEB/8QAHwAAAQUBAQEBAQEAAAAAAAAAAAECAwQFBgcICQoL/8QAtRAAAgEDAwIEAwUFBAQAAAF9AQIDAAQRBRIhMUEGE1FhByJxFDKBkaEII0KxwRVS0fAkM2JyggkKFhcYGRolJicoKSo0NTY3ODk6Q0RFRkdISUpTVFVWV1hZWmNkZWZnaGlqc3R1dnd4eXqDhIWGh4iJipKTlJWWl5iZmqKjpKWmp6ipqrKztLW2t7i5usLDxMXGx8jJytLT1NXW19jZ2uHi4+Tl5ufo6erx8vP09fb3+Pn6/8QAHwEAAwEBAQEBAQEBAQAAAAAAAAECAwQFBgcICQoL/8QAtREAAgECBAQDBAcFBAQAAQJ3AAECAxEEBSExBhJBUQdhcRMiMoEIFEKRobHBCSMzUvAVYnLRChYkNOEl8RcYGRomJygpKjU2Nzg5OkNERUZHSElKU1RVVldYWVpjZGVmZ2hpanN0dXZ3eHl6goOEhYaHiImKkpOUlZaXmJmaoqOkpaanqKmqsrO0tba3uLm6wsPExcbHyMnK0tPU1dbX2Nna4uPk5ebn6Onq8vP09fb3+Pn6/9oADAMBAAIRAxEAPwD4vq5Z1TqSvtD2CS8ooqSgDPu9H+2Vz/2O+s67CqesaPWgFPR7z1Fal5/plc3/AMen4VuWd5QBYooooAKKjorMDQs6uVl2daFaAaFnWhVOzrUs60AuWdXKjqSgAqSiigCOipKr0AV7yq9XKjoNDPqOrFR0AV6KjqOgzJLyo6sV...","name":"a94df035fc91a158c1f9d7a1c769ca66_archive_1580400781.jpg"}],"rotation":"180","now":"Thu Jan 30 16:13:02 2020","motionminarea":250,"name":"Coop Interior","resolution_height":972,"motioncompareframe":"last","is_live":true}`

theyosh, 2020-01-30:
That looks good and what I would expect. So now it is up to you to process it further. Lets hope that will work.

theyosh, 2020-01-30:
And a nice trick with the z-wave! I need to make that also as an emergency kill switch...

Swiftnesses, 2020-01-30:
It's perfect, now I can take automation that step further ;-)

Swiftnesses, 2020-01-30:
Thanks @theyosh, fixed :-)

theyosh, 2020-01-30:
Wicked!

Swiftnesses, 2020-01-30:
Hmmm, so it stopped working, perhaps after I changed the settings for motion (or the pi lost connection?), I rebooted and the webhooks are working again.

BTW, the archive was working throughout (saving images), it's just the webhook that stopped.

Swiftnesses, 2020-01-30:
@theyosh any ideas?

theyosh, 2020-01-30:
What is the screen session telling you? Without it, I have no clue

Swiftnesses, 2020-01-30:
Let me login later, I kicked it and it's been fine since...

theyosh, 2020-03-11:
Hi, not to be rude, but I would like to close this issue. As it has been working for me for months, I have to conclude that the problem is not in my code. So I will close this issue this weekend if there is no response. 

Also the level of response is quite low, so I cannot do anything with this anymore. Please update!

Swiftnesses, 2020-03-11:
Hi @theyosh,

Deeply sorry, it's been a crazy few months - it is mostly working, so we can close.

closed 2020-03-11 by Swiftnesses