theyosh/TerrariumPI #335: Xiaomi Mi Temperature and Humidity Monitor

5point0, 2019-10-09:
Hi theyosh

I have purchased one of these Xiaomi Mi Temperature and Humidity Monitors. Can this connect to your software like the MiFlora?

I have tried to add the bluetooth address and get this error:
2019-10-09 13:02:43,284 - WARNING - terrariumBluetoothSensor - Error getting new data from miflora sensor 'X Temp'. Error message: unpack requires a string argument of length 7.

I really love this software, thanks for all the great work!

theyosh, 2019-10-09:
Hi,

If you are referring to https://www.xiaomistore.pk/mi-bluetooth-temperature-humidity-monitor.html then there is at the moment no support for it.

But we can add it. As there is already a Python library for it. It will probably only work with Python3, so you need to run it on Python 3 to get it stable (if I remember it well).

But we can test it if you have one at home.

5point0, 2019-10-09:
Yea thats it.

I have one close to the PI at the moment, but I'm running Python2 currently.

Where do we start?

theyosh, 2019-10-09:
Ok, if you are depending on this pi, it is maybe not wise to use for testing this. So for you, if you have the time, you can change your setup to Python 3 with the following steps: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-use-python-35

This can take some time, as all python modules will be compiled and installed for python 3. But it should just work. Make sure that when you manual start the software, you add the argument '3' to it.

And update the /etc/rc.local file as stated in the FAQ.

Then I need to add some code. And I am not sure how fast I can do that.... So it could be that you are earlier done with the upgrade then I can add new code.

5point0, 2019-10-09:
So I will bring home a USB card reader tomorrow and clone to my spare SD just incase. 

I will let you know when I'm updated and ready to test tomorrow.

theyosh, 2019-10-09:
Backups are always a wise idea: https://github.com/theyosh/TerrariumPI/wiki/FAQ#which-files-to-backup

theyosh, 2019-10-13:
Hi,

I have made an update. So if you are at Python 3 you could test it with the following steps:

1. `git pull`
2. rerun the installer: `sudo ./install 3` make sure the parameter 3 is there
3. restart
4. add a new sensor. You need to know the bluetooth address for now. Enter this at the sensor address field. You can add the sensor twice for temperature and one for humidity

5point0, 2019-10-17:
Hi, 

I can't seem to install Python 3. I have followed the steps but the installer runs really fast so i don't think its installing.

After sudo ./start.sh 3 i get this in screen -r on loop:

2019-10-17 20:15:24,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.88.233:8090 ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 2, in <module>
    from gevent import monkey, sleep
ImportError: No module named 'gevent'
2019-10-17 20:15:24,000 - INFO    - terrariumWrapper     - Restart counter: 1/5
2019-10-17 20:15:24,000 - INFO    - terrariumWrapper     - Restarting in 10 seconds after running for 0 seconds. Press Ctrl+C now to terminate TerrariumPI.
10 9 8 7 6 5 4 3

theyosh, 2019-10-20:
Hmm strange. It can be that a migrate from 2 to 3 does not always work well.

Then using a second SD card, en directly install it with python3 will solve it. You can just copy settings and other files back to the new version, so it will continue with the same settings and history.

5point0, 2019-10-21:
I tried with a new SD card, flashed stretch but I get the same result when running the sudo ./install.sh 3 command.

I have uploaded an image I got of the errors in the installer, hopefully it will help find the issue.

[image]

theyosh, 2019-10-29:
Hmm,

sorry for the late reply, but it looks like you are missing some packages. Can you manually run:

`sudo apt install python3-pip python3-dev python3-mediainfodll python3-smbus python3-pil python3-opencv python3-numpy python3-lxml` ?

And then rerun the installer?

5point0, 2019-10-30:
So I have run what you suggested above but I got an error `Unable to locate package package python3-opencv`.
I removed `python3-opencv` from the string and everything else installed.
I then tried `sudo pip3 install opencv-python` after some googling and that seemed to work.

I tried the installer again and it seems to have gone a lot better and I was able to start the software with the parameter 3.

I have edited the file `/etc/rc.local` and add then parameter 3 in here but after reboot the software does not start automatically and I need to start it manually. What can the issue be?

I do get readings from the new sensor so the code you added is working. I will keep and eye on it and let you know if all is well in that regard.

5point0, 2019-10-31:
After a while a problem with the sensors updating came about.

`2019-10-30 22:45:40,331 - INFO    - terrariumEnvironment - Update done in 0.01677 seconds. Waiting for 14.98323 seconds for next update
2019-10-30 22:45:40,327 - ERROR   - terrariumEnvironment - Environment temperature sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!
2019-10-30 22:45:40,322 - ERROR   - terrariumEnvironment - Environment humidity sensors are not up to date. Check your sensors on the sensor page. So force the power down to be sure!`

theyosh, 2019-10-31:
Hi,

I just got another issues with python3 and installing it. So I will have a look at the start up issue. It should work.

Your latest log lines does not show any sensors errors. So if this is all you see, then it is very strange.

Could you then try: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

theyosh, 2019-11-10:
I did just a full and clean install with the latest raspian buster image and python3. I do not have any issues. So I would like to ask if it is possible to start over again with a clean install?

Then you can use your current settings file to restore your settings.

5point0, 2019-11-10:
Yes, I have setup another PI with buster and the install went smoothly!
What I have been trying to figure out what error I see as the sensor seems to stop working or doesn't work at all at times.

I get the following so far:

pi@raspberrypi:~/TerrariumPI $ screen -r

   return _write(self._fileno, b)
BrokenPipeError: [Errno 32] Broken pipe
Exception ignored in: <function Peripheral.__del__ at 0x74faa198>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 630, in __del__
    self.disconnect()
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 453, in disconnect
    self._writeCmd("disc\n")
  File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 305, in _writeCmd
    self._helper.stdin.flush()
  File "/usr/local/lib/python3.7/dist-packages/gevent/_fileobjectposix.py", line 159, in write
    return _write(self._fileno, b)
BrokenPipeError: [Errno 32] Broken pipe
2019-11-10 10:14:13,715 - WARNING - terrariumSensor      - Measured value NoneC from mitemp sensor 'Xtemp' is outside valid range 0.00C - 100.00C in 3.15749 seconds.
2019-11-10 10:14:13,721 - WARNING - terrariumBluetoothSensor - Error getting new data from mitemp sensor 'XHum'. Error message: This operation would block forever
        Hub: <Hub '' at 0x75aa8330 epoll default pending=0 ref=0 fileno=4 resolver=<gevent.resolver_thread.Resolver at 0x64c9be30 pool=<ThreadPool at 0x6428c5b0 0/1/10 hub=<Hub at 0x75aa8330 thread_ident=0x1995803344>>> threadpool=<ThreadPool at 0x6428c5b0 0/1/10 hub=<Hub at 0x75aa8330 thread_ident=0x1995803344>> thread_ident=0x76f58ad0>
        Handles:
[]
2019-11-10 10:14:13,723 - WARNING - terrariumSensor      - Measured value None% from mitemp sensor 'XHum' is outside valid range 0.00% - 100.00% in 0.00491 seconds.
HTTPConnectionPool(host='192.168.88.178', port=8090): Max retries exceeded with url: /static/extern/CO2.json (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x6426bc90>: Failed to establish a new connection: [Errno 111] Connection refused'))
2019-11-10 10:14:13,744 - WARNING - terrariumSensor      - Measured value Noneppm from remote sensor 'CO2 test' is outside valid range 0.00ppm - 100.00ppm in 0.01888 seconds.
2019-11-10 10:14:19,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
Expecting value: line 1 column 3 (char 2)
2019-11-10 10:14:23,179 - WARNING - terrariumSensor      - Measured value Noneppm from remote sensor 'CO2 test' is outside valid range 0.00ppm - 10000.00ppm in 0.03551 seconds.
Expecting value: line 1 column 3 (char 2)                                                           
2019-11-10 10:14:49,727 - WARNING - terrariumSensor      - Measured value Noneppm from remote sensor 'CO2 test' is outside valid range 0.00ppm - 10000.00ppm in 0.02905 seconds.
Expecting value: line 1 column 3 (char 2)
2019-11-10 10:15:29,148 - WARNING - terrariumSensor      - Measured value Noneppm from remote sensor 'CO2 test' is outside valid range 0.00ppm - 10000.00ppm in 0.03236 seconds.
Expecting value: line 1 column 3 (char 2)
2019-11-10 10:15:49,736 - WARNING - terrariumSensor      - Measured value Noneppm from remote sensor 'CO2 test' is outside valid range 0.00ppm - 10000.00ppm in 0.03057 seconds.
/usr/local/lib/python3.7/dist-packages/gevent/pywsgi.py:908: ResourceWarning: unclosed file <_io.BufferedReader name='/home/pi/TerrariumPI/log/terrariumpi.log'>
  for data in self.result:
ResourceWarning: Enable tracemalloc to get the object allocation traceback
2019-11-10 10:16:26,112 - WARNING - terrariumBluetoothSensor - Error getting new data from mitemp sensor 'Xtemp'. Error message: 
2019-11-10 10:16:26,114 - WARNING - terrariumSensor      - Measured value NoneC from mitemp sensor 'Xtemp' is outside valid range -9.00C - 60.00C in 6.63709 seconds.
2019-11-10 10:16:35,847 - WARNING - terrariumBluetoothSensor - Error getting new data from mitemp sensor 'XHum'. Error message: Could not read data from Mi Temp sensor 58:2d:34:35:16:da
2019-11-10 10:16:35,849 - WARNING - terrariumSensor      - Measured value None% from mitemp sensor 'XHum' is outside valid range 10.00% - 90.00% in 9.61256 seconds.
Expecting value: line 1 column 3 (char 2)
2019-11-10 10:16:36,002 - WARNING - terrariumSensor      - Measured value Noneppm from remote sensor 'CO2 test' is outside valid range 0.00ppm - 10000.00ppm in 0.03150 seconds.
Expecting value: line 1 column 3 (char 2)
2019-11-10 10:16:53,222 - WARNING - terrariumSensor      - Measured value Noneppm from remote sensor 'CO2 test' is outside valid range 0.00ppm - 10000.00ppm in 0.03176 seconds.

theyosh, 2019-11-11:
Could you remove the CO2 sensor? That is causing a lot of error output. If you want to debug this Xiaomi sensor, make sure that it is the only connected and configured sensor. Else you have much more variable that could case issues.

theyosh, 2019-11-11:
Also, I need the debug from screen -r just after the restart. During startup, there is some scanning and some extra debug. I need that.

So remove all other sensors except the Xiaomi and restart. Then directly look for the text:

start scanning mi temps and Found device type...

That is the first thing I need to know if that works

theyosh, 2019-11-11:
And rerun the installer, because about 20 days ago, the used bluetooth library got updated... maybe that is also helping

theyosh, 2019-11-11:
Did another update. Scanning should now work better. So I hope you can capture the scanning names so I can automatically add the sensors.

5point0, 2019-11-12:
Okay i have removed the co2 sensor and updated.
**Screen -r output:**

     2019-11-12 08:00:23,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.88.178:8090 ...
     2019-11-12 08:00:47,245 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
     2019-11-12 08:00:52,298 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
     start scanning mi temps
     Found device type None
     Found device type LG SH5(DA)
     Found device type None
     Found device type None
     Found device type MJ_HT_V1
     Found device type None
     2019-11-12 08:01:09,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
     /usr/local/lib/python3.7/dist-packages/gevent/pywsgi.py:908: ResourceWarning: unclosed file <_io.BufferedReader name='/home/pi/TerrariumPI/log/terrariumpi.log'>
     for data in self.result:
     ResourceWarning: Enable tracemalloc to get the object allocation traceback

**And this is the log:**

    2019-11-12 08:01:45,848 - INFO    - terrariumSensor      - Updated mitemp sensor 'XHum' from 53.90% to 53.70% in 0.00033 seconds
    2019-11-12 08:01:45,726 - INFO    - terrariumSensor      - Updated mitemp sensor 'Xtemp' from 24.20C to 24.20C in 6.38656 seconds
    2019-11-12 08:01:39,370 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00003 seconds. Waiting for 29.99997 seconds for next update
    2019-11-12 08:01:39,366 - INFO    - terrariumEnvironment - Update done in 0.00266 seconds. Waiting for 14.99734 seconds for next update
    2019-11-12 08:01:39,300 - INFO    - terrariumAudio       - Update done in 0.00002 seconds. Waiting for 29.99998 seconds for next update
    2019-11-12 08:01:24,291 - INFO    - terrariumEnvironment - Update done in 0.00220 seconds. Waiting for 14.99780 seconds for next update
    2019-11-12 08:01:09,579 - INFO    - terrariumEngine      - Update done in 0.29355 seconds. Waiting for 29.70645 seconds for next update
    2019-11-12 08:01:09,438 - INFO    - terrariumSensor      - Updated mitemp sensor 'XHum' from 53.90% to 53.90% in 0.00024 seconds
    2019-11-12 08:01:09,338 - INFO    - terrariumEngine      - Start terrariumPI engine log
    2019-11-12 08:01:09,337 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00000 seconds. Waiting for 30.00000 seconds for next update
    2019-11-12 08:01:09,337 - INFO    - terrariumEngine      - Start terrariumPI webcams
    2019-11-12 08:01:09,311 - INFO    - terrariumSensor      - Updated mitemp sensor 'Xtemp' from 24.20C to 24.20C in 0.00016 seconds
    2019-11-12 08:01:09,285 - INFO    - terrariumEngine      - Start terrariumPI engine
    2019-11-12 08:01:09,285 - INFO    - terrariumAudio       - Update done in 0.00000 seconds. Waiting for 30.00000 seconds for next update
    2019-11-12 08:01:09,284 - INFO    - terrariumEnvironment - Update done in 0.00094 seconds. Waiting for 14.99906 seconds for next update
    2019-11-12 08:01:09,282 - INFO    - terrariumEnvironment - Starting engine
    2019-11-12 08:01:09,280 - INFO    - terrariumWebserver   - Running webserver at :::8090
    2019-11-12 08:01:09,204 - INFO    - terrariumTranslations - Loaded TerrariumPI with 124 translations
    2019-11-12 08:01:09,201 - INFO    - terrariumTranslations - Initialize TerariumPI translations
    2019-11-12 08:01:09,158 - INFO    - terrariumEngine      - TerrariumPI engine is running
    2019-11-12 08:01:09,157 - INFO    - terrariumAudio       - Player loaded with 0 audio files and 0 playlists. Starting engine.
    2019-11-12 08:01:09,138 - INFO    - terrariumEngine      - Done loading terrariumPI webcams. Found 0 webcams in 0.001 seconds
    2019-11-12 08:01:09,137 - INFO    - terrariumEngine      - Loading terrariumPI webcams
    2019-11-12 08:01:09,136 - INFO    - terrariumEnvironment - Done loading terrariumPI Environment 0.001 seconds
    2019-11-12 08:01:09,135 - INFO    - terrariumEnvironment - Loading terrariumPI Environment system
    2019-11-12 08:01:09,133 - INFO    - terrariumEngine      - Done loading terrariumPI doors. Found 0 doors in 0.001 seconds
    2019-11-12 08:01:09,133 - INFO    - terrariumEngine      - Loading terrariumPI doors
    2019-11-12 08:01:09,132 - INFO    - terrariumEngine      - Done starting up terrariumPI switches. Found 0 switches in 5.525 seconds
    2019-11-12 08:01:09,131 - INFO    - terrariumSwitch      - Meross cloud is not enabled.
    2019-11-12 08:01:03,609 - INFO    - terrariumEngine      - Loading previous power switch states from the last 12 hours
    2019-11-12 08:01:03,608 - INFO    - terrariumEngine      - Loaded 0 excluding IDs: []
    2019-11-12 08:01:03,607 - INFO    - terrariumEngine      - Starting up terrariumPI switches
    2019-11-12 08:01:03,607 - INFO    - terrariumEngine      - Done loading terrariumPI sensors. Found 2 sensors in 16.369 seconds
    2019-11-12 08:01:03,606 - INFO    - terrariumSensor      - Updated mitemp sensor 'XHum' from 0.00% to 53.90% in 0.00025 seconds
    2019-11-12 08:01:03,605 - INFO    - terrariumSensor      - Loaded mitemp humidity sensor 'XHum' on location 58:2d:34:35:16:da.
    2019-11-12 08:01:03,603 - INFO    - terrariumSensor      - Updated mitemp sensor 'Xtemp' from 0.00C to 24.20C in 6.24143 seconds
[... cut ...]
    2019-11-12 08:00:45,821 - INFO    - terrariumEngine      - Loading terrariumPI collector
    2019-11-12 08:00:45,821 - INFO    - terrariumEngine      - Done setting terrariumPI authentication
    2019-11-12 08:00:45,819 - INFO    - terrariumEngine      - Setting terrariumPI authentication
    2019-11-12 08:00:45,814 - INFO    - terrariumEngine      - Done Loading terrariumPI config
    2019-11-12 08:00:45,805 - INFO    - terrariumEngine      - Loading terrariumPI config
    2019-11-12 08:00:45,320 - INFO    - terrariumPI          - Starting terrariumPI

5point0, 2019-11-12:
After some time it looks like the mi sensor disconnects but then it doesn't try to reconnect and other sensors also stop updating until reboot.

Screen -r:

    pi@raspberrypi:~ $ screen -r

    return func(*args, **kwargs)
    File "/usr/local/lib/python3.7/dist-packages/btlewrap/bluepy.py", line 64, in disconnect
    self._peripheral.disconnect()
    File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 453, in disconnect
    self._writeCmd("disc\n")
    File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 305, in _writeCmd
    self._helper.stdin.flush()
    File "/usr/local/lib/python3.7/dist-packages/gevent/_fileobjectposix.py", line 159, in write
    return _write(self._fileno, b)
    BrokenPipeError: [Errno 32] Broken pipe
    Exception ignored in: <function BluetoothInterface.__del__ at 0x74fff738>
    Traceback (most recent call last):
    File "/usr/local/lib/python3.7/dist-packages/btlewrap/base.py", line 18, in __del__
    self._backend.disconnect()
    File "/usr/local/lib/python3.7/dist-packages/btlewrap/bluepy.py", line 27, in _func_wrapper
    return func(*args, **kwargs)
    File "/usr/local/lib/python3.7/dist-packages/btlewrap/bluepy.py", line 64, in disconnect
    self._peripheral.disconnect()
    File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 453, in disconnect
    self._writeCmd("disc\n")
    File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 305, in _writeCmd
    self._helper.stdin.flush()
    File "/usr/local/lib/python3.7/dist-packages/gevent/_fileobjectposix.py", line 159, in write
    return _write(self._fileno, b)
    BrokenPipeError: [Errno 32] Broken pipe
    Exception ignored in: <function Peripheral.__del__ at 0x750361e0>
    Traceback (most recent call last):
    File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 630, in __del__
    self.disconnect()
    File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 453, in disconnect
    self._writeCmd("disc\n")
    File "/usr/local/lib/python3.7/dist-packages/bluepy/btle.py", line 305, in _writeCmd
    self._helper.stdin.flush()
    File "/usr/local/lib/python3.7/dist-packages/gevent/_fileobjectposix.py", line 159, in write
    return _write(self._fileno, b)
    BrokenPipeError: [Errno 32] Broken pipe
    /home/pi/TerrariumPI/terrariumSensor.py:105: ResourceWarning: unclosed file <_io.TextIOWrapper name='/dev/null' mode='w' encoding='UTF-8'>
    new_data = self.load_data()
    ResourceWarning: Enable tracemalloc to get the object allocation traceback
    2019-11-12 09:08:51,508 - WARNING - terrariumSensor      - Measured value NoneC from mitemp sensor 'Xtemp' is outside valid range -9.00C - 60.00C in 7.93664 seconds.
    /usr/local/lib/python3.7/dist-packages/gevent/pywsgi.py:908: ResourceWarning: unclosed file <_io.BufferedReader name='/home/pi/TerrariumPI/log/terrariumpi.log'>
    for data in self.result:
    ResourceWarning: Enable tracemalloc to get the object allocation traceback

Log:

    2019-11-12 09:10:43,611 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
    2019-11-12 09:10:43,594 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
    2019-11-12 09:10:28,688 - INFO    - terrariumEnvironment - Update done in 0.00225 seconds. Waiting for 14.99775 seconds for next update
    2019-11-12 09:10:13,673 - INFO    - terrariumEnvironment - Update done in 0.00214 seconds. Waiting for 14.99786 seconds for next update
    2019-11-12 09:10:13,609 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
    2019-11-12 09:10:13,579 - INFO    - terrariumAudio       - Update done in 0.00002 seconds. Waiting for 29.99998 seconds for next update
    2019-11-12 09:09:58,671 - INFO    - terrariumEnvironment - Update done in 0.00218 seconds. Waiting for 14.99782 seconds for next update
    2019-11-12 09:09:43,656 - INFO    - terrariumEnvironment - Update done in 0.00213 seconds. Waiting for 14.99787 seconds for next update
    2019-11-12 09:09:43,608 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
    2019-11-12 09:09:43,564 - INFO    - terrariumAudio       - Update done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
    2019-11-12 09:09:28,654 - INFO    - terrariumEnvironment - Update done in 0.00217 seconds. Waiting for 14.99783 seconds for next update
    2019-11-12 09:09:13,639 - INFO    - terrariumEnvironment - Update done in 0.00223 seconds. Waiting for 14.99777 seconds for next update
    2019-11-12 09:09:13,607 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
    2019-11-12 09:09:13,548 - INFO    - terrariumAudio       - Update done in 0.00002 seconds. Waiting for 29.99998 seconds for next update
    2019-11-12 09:08:58,636 - INFO    - terrariumEnvironment - Update done in 0.00096 seconds. Waiting for 14.99904 seconds for next update
    2019-11-12 09:08:51,508 - WARNING - terrariumSensor      - Measured value NoneC from mitemp sensor 'Xtemp' is outside valid range -9.00C - 60.00C in 7.93664 seconds.
    2019-11-12 09:08:51,455 - WARNING - terrariumBluetoothSensor - Error getting new data from mitemp sensor 'Xtemp'. Error message: [Errno 32] Broken pipe
[... cut ...]
    2019-11-12 09:08:13,950 - INFO    - terrariumEngine      - Update done in 0.42970 seconds. Waiting for 29.57030 seconds for next update
    2019-11-12 09:08:13,801 - INFO    - terrariumSensor      - Updated script sensor 'Co2 test' from 394.00ppm to 394.00ppm in 0.00034 seconds
    2019-11-12 09:08:13,675 - INFO    - terrariumSensor      - Updated mitemp sensor 'XHum' from 51.50% to 51.50% in 0.00031 seconds
    2019-11-12 09:08:13,613 - INFO    - terrariumEnvironment - Update done in 0.00214 seconds. Waiting for 14.99786 seconds for next update

theyosh, 2019-11-13:
Hmm, not sure why. If a sensor is failing, it should not crash the rest of the system. As it looks like doing something with gevent, that could have some strange effects.

It could be that the Python library is not that stable, or not 100% thread safe... but that is something I cannot fix.

I did made some new changes. It can now autodetect sensors during startup. And I did try to clear the bleutooth resources after a measurement so that the error above would not happen. But I can't be sure. As I can't test it :(

Do a `git pull` and restart. Lets see what happens

5point0, 2019-11-14:
Okay I have updated and Rebooted and everything seems to be working and I can see the sensor was automatically added, I have added all the logs below. I will try take the sensor further away and see what it does.

Screen -r:

    2019-11-14 21:59:54,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.88.178:8090 ...
    2019-11-14 22:00:20,372 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
    2019-11-14 22:00:25,425 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
    2019-11-14 22:00:37,891 - WARNING - terrariumSensor      - Measured value 382.0ppm from script sensor 'Co2 test' is outside valid range 0.00ppm - 100.00ppm in 2.11872 seconds.
    2019-11-14 22:00:43,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
    /usr/local/lib/python3.7/dist-packages/gevent/pywsgi.py:908: ResourceWarning: unclosed file <_io.BufferedReader name='/home/pi/TerrariumPI/log/terrariumpi.log'>
    for data in self.result:
    ResourceWarning: Enable tracemalloc to get the object allocation traceback

Log:

    2019-11-14 22:01:28,650 - INFO    - terrariumEnvironment - Update done in 0.00229 seconds. Waiting for 14.99771 seconds for next update
    2019-11-14 22:01:22,157 - INFO    - terrariumEngine      - Update done in 8.58025 seconds. Waiting for 21.41975 seconds for next update
    2019-11-14 22:01:22,004 - INFO    - terrariumSensor      - Updated script sensor 'Co2 test' from 382.00ppm to 382.00ppm in 1.86749 seconds
    2019-11-14 22:01:20,017 - INFO    - terrariumSensor      - Updated mitemp sensor 'XHum' from 33.20% to 33.30% in 0.00027 seconds
    2019-11-14 22:01:19,895 - INFO    - terrariumSensor      - Updated mitemp sensor 'Xtemp' from 25.50C to 25.40C in 0.00028 seconds
    2019-11-14 22:01:19,764 - INFO    - terrariumSensor      - Updated mitemp sensor '' from 33.20% to 33.30% in 0.00029 seconds
    2019-11-14 22:01:19,641 - INFO    - terrariumSensor      - Updated mitemp sensor '' from 25.50C to 25.40C in 6.02619 seconds
    2019-11-14 22:01:13,645 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00003 seconds. Waiting for 29.99997 seconds for next update
    2019-11-14 22:01:13,642 - INFO    - terrariumEnvironment - Update done in 0.00263 seconds. Waiting for 14.99737 seconds for next update
    2019-11-14 22:01:13,576 - INFO    - terrariumAudio       - Update done in 0.00002 seconds. Waiting for 29.99998 seconds for next update
    2019-11-14 22:00:58,573 - INFO    - terrariumEnvironment - Update done in 0.00219 seconds. Waiting for 14.99781 seconds for next update
    2019-11-14 22:00:46,024 - INFO    - terrariumEngine      - Update done in 2.46289 seconds. Waiting for 27.53711 seconds for next update
    2019-11-14 22:00:45,869 - INFO    - terrariumSensor      - Updated script sensor 'Co2 test' from 0.00ppm to 382.00ppm in 1.77958 seconds
    2019-11-14 22:00:43,957 - INFO    - terrariumSensor      - Updated mitemp sensor 'XHum' from 33.20% to 33.20% in 0.00013 seconds
    2019-11-14 22:00:43,838 - INFO    - terrariumSensor      - Updated mitemp sensor 'Xtemp' from 25.50C to 25.50C in 0.00013 seconds
    2019-11-14 22:00:43,714 - INFO    - terrariumSensor      - Updated mitemp sensor '' from 33.20% to 33.20% in 0.00024 seconds
    2019-11-14 22:00:43,614 - INFO    - terrariumEngine      - Start terrariumPI engine log
    2019-11-14 22:00:43,614 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00000 seconds. Waiting for 30.00000 seconds for next update
    2019-11-14 22:00:43,613 - INFO    - terrariumEngine      - Start terrariumPI webcams
    2019-11-14 22:00:43,589 - INFO    - terrariumSensor      - Updated mitemp sensor '' from 25.50C   to 25.50C in 0.00018 seconds
    2019-11-14 22:00:43,560 - INFO    - terrariumEngine      - Start terrariumPI engine
    2019-11-14 22:00:43,560 - INFO    - terrariumAudio       - Update done in 0.00000 seconds. Waiting for 30.00000 seconds for next update
    2019-11-14 22:00:43,559 - INFO    - terrariumEnvironment - Update done in 0.00095 seconds. Waiting for 14.99905 seconds for next update
    2019-11-14 22:00:43,558 - INFO    - terrariumEnvironment - Starting engine
    2019-11-14 22:00:43,555 - INFO    - terrariumWebserver   - Running webserver at :::8090
    2019-11-14 22:00:43,496 - INFO    - terrariumTranslations - Loaded TerrariumPI with 124 translations
    2019-11-14 22:00:43,493 - INFO    - terrariumTranslations - Initialize TerariumPI translations
    2019-11-14 22:00:43,451 - INFO    - terrariumEngine      - TerrariumPI engine is running
    2019-11-14 22:00:43,449 - INFO    - terrariumAudio       - Player loaded with 0 audio files and 0 playlists. Starting engine.
    2019-11-14 22:00:43,431 - INFO    - terrariumEngine      - Done loading terrariumPI webcams. Found 0 webcams in 0.001 seconds
    2019-11-14 22:00:43,430 - INFO    - terrariumEngine      - Loading terrariumPI webcams
    2019-11-14 22:00:43,429 - INFO    - terrariumEnvironment - Done loading terrariumPI Environment 0.001 seconds
    2019-11-14 22:00:43,428 - INFO    - terrariumEnvironment - Loading terrariumPI Environment system
    2019-11-14 22:00:43,427 - INFO    - terrariumEngine      - Done loading terrariumPI doors. Found 0 doors in 0.001 seconds
[... cut ...]
    2019-11-14 22:00:17,451 - INFO    - terrariumEngine      - Loading terrariumPI collector
    2019-11-14 22:00:17,450 - INFO    - terrariumEngine      - Done setting terrariumPI authentication
    2019-11-14 22:00:17,448 - INFO    - terrariumEngine      - Setting terrariumPI authentication
    2019-11-14 22:00:17,443 - INFO    - terrariumEngine      - Done Loading terrariumPI config
    2019-11-14 22:00:17,433 - INFO    - terrariumEngine      - Loading terrariumPI config
    2019-11-14 22:00:16,925 - INFO    - terrariumPI          - Starting terrariumPI

5point0, 2019-11-14:
It looks like its behaving correctly now when I take the sensor away:

    2019-11-14 22:22:02,558 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
    2019-11-14 22:22:02,556 - INFO    - terrariumEnvironment - Update done in 0.00228 seconds. Waiting for 14.99772 seconds for next update
    2019-11-14 22:22:02,455 - INFO    - terrariumAudio       - Update done in 0.00002 seconds. Waiting for 29.99998 seconds for next update
    2019-11-14 22:21:47,553 - INFO    - terrariumEnvironment - Update done in 0.00221 seconds. Waiting for 14.99779 seconds for next update
    2019-11-14 22:21:46,567 - WARNING - terrariumSensor      - Measured value NoneC from mitemp sensor 'AutoTemp' is outside valid range 1.00C - 40.00C in 14.07544 seconds.
    2019-11-14 22:21:46,565 - WARNING - terrariumBluetoothSensor - Error getting new data from mitemp sensor 'AutoTemp'. Error message:
    2019-11-14 22:21:32,555 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
    2019-11-14 22:21:32,551 - INFO    - terrariumEnvironment - Update done in 0.00241 seconds. Waiting for 14.99759 seconds for next update
    2019-11-14 22:21:32,451 - INFO    - terrariumAudio       - Update done in 0.00002 seconds. Waiting for 29.99998 seconds for next update
    2019-11-14 22:21:19,535 - INFO    - terrariumEngine      - Update done in 17.08527 seconds. Waiting for 12.91473 seconds for next update
    2019-11-14 22:21:19,385 - INFO    - terrariumSensor      - Updated script sensor 'Co2 test' from 391.00ppm to 528.00ppm in 1.89245 seconds
    2019-11-14 22:21:17,549 - INFO    - terrariumEnvironment - Update done in 0.00287 seconds. Waiting for 14.99713 seconds for next update
    2019-11-14 22:21:17,370 - WARNING - terrariumSensor      - Measured value None% from mitemp sensor 'AutoHum' is outside valid range 10.00% - 100.00% in 7.00869 seconds.
    2019-11-14 22:21:17,368 - WARNING - terrariumBluetoothSensor - Error getting new data from mitemp sensor 'AutoHum'. Error message:
    2019-11-14 22:21:10,238 - WARNING - terrariumSensor      - Measured value NoneC from mitemp sensor 'AutoTemp' is outside valid range 1.00C - 40.00C in 7.75077 seconds.
    2019-11-14 22:21:10,236 - WARNING - terrariumBluetoothSensor - Error getting new data from mitemp sensor 'AutoTemp'. Error message: Could not read data from Mi Temp sensor 58:2d:34:35:16:da
    2019-11-14 22:21:02,554 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update
    2019-11-14 22:21:02,546 - INFO    - terrariumEnvironment - Update done in 0.00232 seconds. Waiting for 14.99768 seconds for next update
    2019-11-14 22:21:02,448 - INFO    - terrariumAudio       - Update done in 0.00002 seconds. Waiting for 29.99998 seconds for next update
    2019-11-14 22:20:47,544 - INFO    - terrariumEnvironment - Update done in 0.00220 seconds. Waiting for 14.99780 seconds for next update
    2019-11-14 22:20:32,874 - INFO    - terrariumEngine      - Update done in 0.42997 seconds. Waiting for 29.57003 seconds for next update
    2019-11-14 22:20:32,725 - INFO    - terrariumSensor      - Updated script sensor 'Co2 test' from 391.00ppm to 391.00ppm in 0.00040 seconds
    2019-11-14 22:20:32,602 - INFO    - terrariumSensor      - Updated mitemp sensor 'AutoHum' from 33.70% to 33.70% in 0.00032 seconds
    2019-11-14 22:20:32,552 - INFO    - terrariumEngine      - Webcam update(s) done in 0.00001 seconds. Waiting for 29.99999 seconds for next update

I will bring the sensor close enough and let it run and see if everything is working okay.

5point0, 2019-11-15:
Im not sure why but I'm getting a couple of these type of warnings:

    2019-11-15 08:37:01,232 - WARNING - terrariumEngine      - Updating took to much time. Needed 34172.23723 seconds which is 34142.23723 more then the limit 30

Seems it stops scanning for sensors for some time. Oddly enough it corrected itself when I opened the webpage to view the log.

theyosh, 2019-11-18:
Hmm, this sounds like some bluetooth issues. I have it also with the other miflora sensors. And I get the idea that new batteries will solve it for a period.

Also, I have a custom cron job, that restarts my bluetooth system on my Pi to make sure it does not get stuck....

5point0, 2019-11-19:
So it seems when the device isn't super close to the PI then some readings are none but then the software only checks again when I load the webpage. So if I close the webpage it doesn't take readings of any sensors but when I open the webpage it starts reading everything again. 

If I just stay on the log in the webpage it doesn't stop reading the sensor even if I get a none from the sensor. So it's only when i close it.

I hope that makes sense.

theyosh, 2019-11-19:
No, this does not make sense :P Your explanation is clear. But I have no clue why this is happening.

It looks like that it could be caused by the bleutooth library that I am using. And the hard part is that I do not have such device... So, I need some time to think this through

theyosh, 2019-11-19:
Is this the only bluetooth device you are using? Could you describe a bit of your setup?

5point0, 2019-11-19:
I have basically setup a testing PI, so it only has this sensor and the Co2 sensor.

Would giving you access to the setup help?

theyosh, 2019-12-07:
Hmm, rewritten some part. Now it should keep the bluetooth connection open... see if that will make a difference... could you do a `git pull` and restart?

vacuumboots, 2019-12-18:
I have a xiaomi Mi Temperature and Humidity Monitor. 
I installed a brand new raspbian lite image, and then installed a fresh version of TerrariumPI. 
I'm using a 3b+ and installed the python3 version. 
The sensors (temp and humidity) were picked up immediately and correctly. 
The reporting/ logging from the device seems to be stable for at least 24 hours. 
I also tried taking the sensor out-of-range for 5 minutes, and then bringing it back.
The TerrariumPi software gracefully handled the sensor being gone and brought back. 

All this to say the Xiaomi Mi Temperature and Humidity Monitor seems to be working really well in my setup.

theyosh, 2019-12-18:
@vacuumboots NICE! Thanks for testing and reporting back!

I do know that I also have some issues with my bluetooth on the pi. So I have a cron job that restarts the bluetooth service every hour... sometimes it get stuck in reading the data...

But that is different then getting no data... So then for now, it looks like it is working like expected. And will leave the new code in place.

@5point0 If you can also test it, we know for sure that it works

victorhooi, 2019-12-29:
FYI - there seems to be version 2 of the Xiaomi Temperature and Humidity sensor out now:

https://www.gizmochina.com/2019/11/11/mijia-bluetooth-thermometer-hygrometer-2-launched-with-new-features-for-%C2%A549-7/

theyosh, 2020-01-18:
Closing as it is working, and no response

For new hardware types, open a new ticket.

closed 2020-01-18 by theyosh

5point0, 2020-01-19:
Hi there, sorry I was busy with a revamp of the setup!
So far so good, getting good readings all the time now.

5point0, 2020-02-05:
Hi @theyosh 

Just for reference, how many of these do you think one can add?

theyosh, 2020-03-17:
@5point0 I have only tested with 3 bluetooth MiFlora devices. Those a bit different, but that was not a problem. I think you can use as much as the bluetooth specifications allows. But I do not know how much that is.

albertopinguino, 2020-08-31:
Hi,
I have 4 XIAOMI sensors installed... 3like this:
https://it.aliexpress.com/item/4001098178650.html?spm=a2g0o.productlist.0.0.60ce6ef9Xu61Xy&algo_pvid=d25a66c9-6d36-49c3-abbc-80b3b32bbf92&algo_expid=d25a66c9-6d36-49c3-abbc-80b3b32bbf92-13&btsid=0b0a0ac215988797166503628e1271&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
And 1 like this:
https://it.aliexpress.com/item/4001127122070.html?spm=a2g0o.productlist.0.0.60ce6ef9Xu61Xy&algo_pvid=d25a66c9-6d36-49c3-abbc-80b3b32bbf92&algo_expid=d25a66c9-6d36-49c3-abbc-80b3b32bbf92-58&btsid=0b0a0ac215988797166503628e1271&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_
The work well for about 1 hour... after this, watching the LOG, all is stuck. If I do a "Scan BluetoothLE", all is working back to normal. It is possible to schedule a cron with a command to do the same thing every hour? If so, what is the correct comand?
Thank you in advance!

theyosh, 2020-08-31:
Hi,

I have a crontab running every hour that restarts the bluetooth stack. In my pi crontab:

`5 * * * * sudo service bluetooth restart`

So 5 min past every hour, my bluetooth is restarted. And there is a change that the bluetooth is restarted and that my software is trying to use it. That is not a problem, as 30 seconds later a new attempt is made. This does solve the issue for me.

I do not know why the bluetooth is hanging sometimes.... Maybe I am using it wrongly in my code... but then again, I use 3rd party libraries..

albertopinguino, 2020-08-31:
Thank you! I'll let you know if it is working...

albertopinguino, 2020-09-01:
No, not working. Still stops working. Only solutions are selecting "Scan Bluetooth LE" from web or restarting RPIEasy service or reboot the raspberry...

theyosh, 2020-09-01:
Hm, I  do not know how the bluetooth devices will react if multiple software system want to use them. So I would not run RPIEasy with my software. As the will both read out the same sensors, and that can cause collisions.

So, I can only advice not to use RPIEasy if you want to use my software.

theyosh, 2020-09-01:
Or, does RPIEasy provide an API where you can read out the sensors through a script? Then you could use the remote sensors type, so that the sensors are only access by 1 software system.