theyosh/TerrariumPI #336: Manual mode for switches resets after power outage

duenni, 2019-10-20:
Hi,

I had a switch in manual mode when a power outage occured, after the Raspberry rebooted and TerrariumPi started again the manual mode on the switch was gone. The switch returned to the timer set on it and switched to "on" state.
The switch should remain in manual mode after a power outage.

Regards

theyosh, 2019-10-31:
Hmm, that sounds like a bug.... will investigate it. Thanks for reporting!!

theyosh, 2019-11-10:
The manual mode is not store in the settings, as it is a temporary state. So normally, you would use the manual mode for just a quick action that needs to be done.

So the state is not stored and will not survive a reboot.

Now the state is saved, but I cannot guarantee that the power state will be restored/stays the same way during a reboot. There are more factors that influence the state of the power switch.

After a `git pull` do a restart and toggle the manual mode off and then on again. Then the state should be saved and survive a reboot.

closed 2019-12-02 by theyosh