theyosh/TerrariumPI #337: Using weather for temperatures

kebs19, 2019-10-24:
Hello there! 
I'm using Terrariumpi in a 76 litres aquarium, trying to mimick an Amazon River environment. It's nice to have the lights turning on and off at the same time as the Amazon, but it would be great if I can get the temperatures from the weather and apply then to the water. 
I know that the weather forecast gives temperatures in the air and that would require some offset...
I live in Brazil but the Amazon weather is waaaay hotter than where I live (tbh we even have different time zones), that's why a dynamic temperature range would be nice to have. 

Another nice feature would be able to use a different temperature sensor (a dht11 one) inside the aquarium cover that could trigger a fan if the LEDs are getting too hot. I know I can use the average temperature to trigger the fan, but the LEDs can get too hot faster than the underwater sensor can detect. 

Thanks

theyosh, 2019-10-31:
Hi,

I put it on the wish list :P At the moment I am not doing much updates due to lack of time.

Your weather mirroring wish will be hard, and maybe never implemented. But multiple temperature zones with custom power actions is something I like also. So that is a bit higher on the wish list.

theyosh, 2021-09-01:
Hi,

it took some time, but with TerrariumPI 4 I think this is now possible. You van have multiple temperature zones in 1 enclosure.

and you can add a new temperature sensor that is of type `remote`. And use openweathermap.org as remote source according to: <https://theyosh.github.io/TerrariumPI/faq/remote_data/>
You can now use an offset (negative or positive) to the air temperature value. And then you can use another temperature zone in your enclosure that heats the water.

theyosh, 2021-12-10:
Closing this issue, as with the new [release 4.1.0](https://github.com/theyosh/TerrariumPI/releases/tag/4.1.0) this is now possible.

closed 2021-12-10 by theyosh