theyosh/TerrariumPI #339: DHT22 Sensor issue

polabear33, 2019-10-26:
I've seen a few things about issues with DHT22 sensors. However, I already have these available to test with. I may look into upgrading in the future. 

I cannot get the sensors to read anything on the dashboard. I have attached how I have them entered as well as the debug screen

Please help! I've been trying to figure this out for hours

Thank you!

[image]
[image]

AndrejWeber, 2019-10-26:
You have to enter the physical PIN number in address field. 
If you want to use gpio27, you have to use address 36.
Address 27 is used for i2c bus

polabear33, 2019-10-27:
Where do you get 36 from? From what I'm seeing it should be 13. But I have
one labeled 13 that isn't working. 13 isn't receiving data either. What am
I missing to see 36?

theyosh, 2019-10-27:
Hi,

wire your hardware according https://github.com/theyosh/TerrariumPI/wiki/Hardware#gpio-pin-setup and https://github.com/theyosh/TerrariumPI/wiki/Hardware#dht22

But you image shows a different error. The problem looks like you have my software running multiple times. Or something else is running on port :8090.

So do a reboot? And then see how it works again?

polabear33, 2019-10-27:
I checked on it today and it appears 13 is now working. I will adjust them all to the physical pin numbers. Thank you! 

I have rebooted but am unsure if it's running multiple on port :8090. When I do a screen -r I do have two things pull up. I am unsure how to kill one of them.

polabear33, 2019-10-27:
It looks like I was able to close it! Thanks for all the help. It looks like my sensors are still pulling a null value after updating, but it might just need more time before it pulls data in. 

Here's an updated debug - My values will be out of range. I do not have heat nor humidity currently running to raise the values. I was originally using RF to run them but I didn't see that as an option to run under switches. I'll likely be switching those over to a relay. If you have any recommendations for that, I'd be appreciative 
[image]

theyosh, 2019-10-31:
Hi, for power switches here is a list of supported hardware: https://github.com/theyosh/TerrariumPI/wiki/Hardware#power-switches

Only Denkovi is a bit harder to work with. Also, this RF is supported: https://energenie4u.co.uk/res/pdfs/ENER314%20UM.pdf

Also, when your sensors are measuring outside your setup values, other parts of the system will also fail and start giving errors. So it would suggest to at least configure your sensors at a very wide range and then tightening it when it is running.

theyosh, 2019-11-10:
Can this issue being closed? Are there still issues remaining

closed 2019-11-11 by polabear33