theyosh/TerrariumPI #340: Fresh Install not starting

Haidy777, 2019-10-29:
Hello,
I created an SD Card with a fresh raspbian buster lite image and installed TerrariumPI with Python 3.
When it tries to start it get the following error. I couldn't find a similar case in the issues.
I'm using a Model B V1.1.

```
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 21, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 33, in <module>
    from terrariumWebcam import terrariumWebcam, terrariumWebcamSourceException
  File "/home/pi/TerrariumPI/terrariumWebcam.py", line 11, in <module>
    import cv2
ImportError: /usr/lib/arm-linux-gnueabihf/libvtkCommonDataModel-6.3.so.6.3: unexpected reloc type 0x03
```

theyosh, 2019-10-29:
Hm,

this is strange. How big is the SD card? Is there still some free space left?

And you could try to rerun the installer once more. Maybe something went wrong, and will be fixed a second time.

Haidy777, 2019-10-30:
It's a 32 GB SD card, with the root partition sitting at 12% usage.
I've tried to let the installer run a second time, it ran through without any problems, like the first time, but the problem still arises.
Could it have something to do with the new debian version?

theyosh, 2019-10-31:
Hmm, strange.

At this moment I have no clue. So I need to test this with a new setup. So that will take some time. But it is strange. I have it running on Buster on my develop version, and it was installing correctly.

But I have seen that sometimes packages are changing dependencies, and I need to add those to the installer script. So I need to test this

Haidy777, 2019-11-01:
Thank you. If I can help you somehow, let me know :)

theyosh, 2019-11-10:
Hm, I just did a fully clean install, and no issues. What kind of Pi do you use?

I have a:
Raspberry Pi 3 Model B Rev 1.2

Haidy777, 2019-11-10:
Interesting, I think I've got the same. I will try again tomorrow and keep you updated.

theyosh, 2019-12-02:
Hi, as there is no update, I expect it is working?

closed 2020-01-18 by theyosh