theyosh/TerrariumPI #342: Dht11 not working

kebs19, 2019-11-14:
Hello!
I'm using a dht11 sensor at Port 37 and Terrariumpi keeps saying that the read values are "none%".
When I use the dht11 library from Adafruit it gives me the right values, using the 'sudo python simpletest.py". 
But inside Terrariumpi it keeps saying 'none%' or 'null'.
Considering it's working with adafruit dht11 library, how can I make it work with Terrariumpi?

theyosh, 2019-11-18:
This is a user issue. With `sudo` everything works :p

What happens when you run it without sudo? Does it still give the right output? And is it really a DHT11? As I only have tested the DHT22

kebs19, 2019-11-18:
Hey there :)
When I run it without sudo, it hangs for a minute than gives an error. 

pi@raspberrypi:~/Adafruit_Python_DHT/examples $ sudo python simpletest.py
Temp=21.0*C  Humidity=76.0%

pi@raspberrypi:~/Adafruit_Python_DHT/examples $ python simpletest.py
Failed to get reading. Try again!

(When I'm running as root (with "sudo su") it works without the 'sudo' part)

kebs19, 2019-11-19:
As a workaround I reinstalled using the root user. 
(gave sudo a password and allowed it to login via ssh).
Now it works :)

closed 2020-01-18 by theyosh