theyosh/TerrariumPI #343: OLED 1309 display working erratically

kebs19, 2019-11-14:
I use a small 1309 OLED display in my Terrariumpi.
But it works very erratically. Sometimes it hangs at the picture of my fish for hours, and another times it gives a warning message about a non functional sensor, or a power switch changed state (even if I set it up to show only the "warning" and "errors" message).
That informative message about the temperature, time online, etc, is almost impossible to show. 
Since it's an OLED display, a static image will burn the display very soon. 
How can I make it show the stats message?

theyosh, 2019-11-18:
Hmm, it should update every 30 seconds... so that is strange.

Also, the support of the oled screen is a bit limited, as it is small and I am not a well UX designer. So that is I just made it to update every 30 seconds. But still, I think it is not easy to read. So maybe it should somehow show a summary?

The problem is that it is a small screen, and that I have to take that in account. So I can't just make long sentences and limited to 4 a 5 lines at once..

kebs19, 2019-11-19:
I`m experimenting with different typefonts and I'll get you informed.
I think this behavior is related to #342 since I could make the sensor work the display is working fine by now.
I just set up the system, I'll wait a few more days before calling it done

theyosh, 2020-01-28:
Hi, any updates on this?

kebs19, 2020-01-28:
Hey there! I saw that the square around the text (which is static) was burning the Oled screen... 
So I changed the system for a 16-char LCD screen.

closed 2020-01-28 by kebs19