theyosh/TerrariumPI #344: Temperature control not working properly

jeff26300, 2019-11-17:
Hello,

I am completely new on the forum and I need some help for the temperature control.
First, I would like to say that the interface looks great and all my congratulations for the nice work!
I have recently set up TerrariumPi to convert a Biopod Aqua2 to Raspberry, because its PCB is definitely not reliable. I have 5 frogs and a couple of gekos. I have been exploring the interface for few days and I managed to first control the ground temperature of the vivarium with a 1Wire thermo sensor and by connecting the original heater on a relay (via GPIO 11). The loop works but I am not able to fine tune the control.

The terrarium needs a temp of 24 in the day and 20 in the night, then I set up the heating based on the light timing (see pictures attached) and the sensor with alarms at 23.8 and 24.2 for the day, which will drop by 4 degC in the night. However, as you can see on the picture, the heater starts every hour for a period of 10min and does not work until the first alarm is reached. I decided to put the Power On for 600s (10 min) to understand the reaction of the system, but I want it to heat until the temperature is within the 2 alarms. Also I don't understand where the 1h comes from. Can you assist me on this control?

Cheers,

[image]
[image]
[image]

theyosh, 2019-11-18:
Hmm, it could be that the jump in temperature is to big. But that is just a thought.

So for now, I think putting the power on value to 0 will put on the power for ever, until max temp is reached. So could you try that. Other settings looks good.

jeff26300, 2019-11-23:
Thank you! Changing the value to 0 definitely helped but it was apparently not the single cause. The terrarium is currently set on "Sensor" so the "Timer On" field is by default greyed out. However, because I tried several parameters before, I left in this filed the value 405 (picture 1), and I didn't put it to 0 because I thought it would not be taken in consideration when not activated. However, it is! Since I activated and set it back to 0s, with the mode on "Timer", the "Sensor" mode now works perfectly, as you can see on the second picture. I don't know if this is a bug but the greyed out value field seems to be used for the control.
[image]
[image]

I also would like to share some pictures of our terrarium:

[image]

![IMG_3099](https
[image]
://user-images.githubusercontent.com/57871568/69476546-7a5aee80-0ddb-11ea-90de-19c42e8308d5.jpg)

Thank you for making this nice program available!

Jeff

theyosh, 2019-12-02:
Hi,

thanks for the awesome photos. Those are poisoned frogs? If you have some extra time, could you post them at #210 ?

And about your temperature issue, I cannot replicated it. Here it works like expected even when there are non zero values in the greyed out fields. Those values will not being used.

So, I do not know why you have this issue. Is should not be one.

closed 2020-01-18 by theyosh

jeff26300, 2020-05-05:
Hello,
I am back again on a temperature control issue. I still have the same set up as described in issue #344 and usually everything works fine. But time to time the heater is not controlled by the sensor and exceeds the alarm value and never stops (see picture). Some days it goes very well and other days we have to stop the heating in order not to get fried frogs... What we found also strange is that time to time the button of the heater switch appears blue (off) and its graph shows power running, which is true because the temperature increases. Can it be related to a full history log that I need to purge? Or the CPU overload described in issue #384 ?

[image]

Cheers