theyosh/TerrariumPI #345: DHT22 not getting the data on pi zero

Died115, 2019-11-30:
Hello, Since i have Multiple terrariums running Theyosh TerrariumPi. (Great work btw, in the past everything ran off 32 bit PIC microcontrollers nothing as fancy as this software)
I started looking into the possibility to use the Pi zero to reduce cost.
I did notice however that the software is not able to get the data from a DHT22 on a Pi zero.
I used an oscilloscope to monitor the signal. There is no capacitive loading of the signal nor a lot of interference. With the standard Adafruit DHT22 library i was able to pull the data.
I installed the software using the super user command: "sudo ./install.sh" 
Also started the software with the super user command.
The I2C interface/SPI interface/1 wire interface are also enabled using raspi-config.
Any Idea what the problem could be?

theyosh, 2019-12-02:
I guess you have the same issue as #342 . Somehow you need to be root to read out the GPIO pins as it looks like.

Not sure why, but you could try to change the start.sh file and change the user pi to user root and restart it. Then everything runs as root. That is a bit of security risk.. but as long you do not put it online, it is fine

Died115, 2019-12-02:
Thanks for the quick reply.
I'll try it.

theyosh, 2019-12-05:
Could you also try this and paste the results here:

`ls -l /dev/gpiomem`

Died115, 2019-12-15:
[image]
 there you go

Died115, 2019-12-16:
Installing everything in the /root folder and then running everything as root solved the problem for me. Now my Pi zero is getting the DHT22 values. :)

theyosh, 2020-01-18:
Ok, on Pi Zero it has to be run as root....

closed 2020-01-18 by theyosh

RaMcHiPTech, 2020-04-06:
How did you install it all in the root folder?

> Installing everything in the /root folder and then running everything as root solved the problem for me. Now my Pi zero is getting the DHT22 values. :)

How did you install it in the root folder?

Died115, 2020-04-06:
Using the command:
sudo -i
Then you clone from git and do the install.