theyosh/TerrariumPI #346: Update Gentella to Bootstrap 4 version?

victorhooi, 2019-12-08:
I know it's still in beta however, Gentella appears to have a new Bootstrap 4-based release:

https://github.com/ColorlibHQ/gentelella/releases/tag/2.0-beta1
https://github.com/ColorlibHQ/gentelella/releases/tag/2.0-beta2

Would it be worth testing these out yet, to see if there's any changes needed with TerrariumPi? (e.g. in another branch)

theyosh, 2019-12-08:
Hi,

yes I saw it and tested it. Does not work instantly. Some HTML looks a bit changed. So for now I have the installer fixed at 1.4.0 version. And I will try to make it work with version 2.0

theyosh, 2019-12-08:
Also, have you had time to check #325 ? There are systemd scripts in the contrib folder

victorhooi, 2019-12-13:
I'm going to try it out - looking to get a new RPI4, and will try on that =).

theyosh, 2020-06-24:
Closing due to not much activity on the Gentella repository.

closed 2020-06-24 by theyosh

victorhooi, 2020-12-27:
The Gentella repository seems to have been fairly active in the last two years:

https://github.com/ColorlibHQ/gentelella/commits/master

I believe we're still based against the January 2017 version.

Do you think it's worth rebasing against current Gentella yet?

theyosh, 2020-12-27:
Hi,
I am not upgrading Gentella. As I am started to rewrite the software for version 4, I started to use a different interface

https://appseed.us/admin-dashboards/flask-dashboard-adminlte / https://flask-dashboard-adminlte.appseed.us/index

As I think there not much progress at Gentella. 50 commits in 2 years is not much in my opinion. Also the demo site they have is still broken. So I decided to switch to a different template.

victorhooi, 2020-12-30:
Got it - yeah, if the Gentella project isn't active enough, I can see that being an issue in terms of maintenance.

I suppose vanilla Bootstrap wouldn't be enough right? As in, you need additional "admin backend" style components?

Or alternatively, would something like [Tailwind UI](https://tailwindui.com/) work for the TerrariumPi use case?

(I am happy to donate the cost of the Tailwind UI license to the project).