theyosh/TerrariumPI #348: FTDI relay states

vinnio, 2020-01-04:
FTDI relays are succesfully controlled using the switches. However, when one of the relays is switched on using ./drcontrol.py for instance, the state is not updated in the webinterface. After inspection of TerrariumSwitch.py, I haven't come accross an section/def "get_hardware_state" for FTDI devices. 
Is this possible to include?

theyosh, 2020-01-18:
Done! do a git pull and it should work....

theyosh, 2020-01-28:
Have you been able to test this?

vinnio, 2020-01-29:
I have tested it. Unfortunately, relay's 1,2,3 and 4 seems to work but relays 5,6,7 and 8 shows some strange behavior. When relay 8 is turned on using a command (./drcontrol.py -d DAE001XA -r 8 -c ON), TerrariumPi turns on Relay 7 and shows relay 7 as active, will relay 8 is shown deactivated.
After some time, the page is updated, relay 6 is turned on (TerrariumPi, no interference from my part), relays 7 and 8 are still activated, but only 7 and 6 are shown to be active. 

After some time again, relay 5 is activated, showing relays 5, 6 and 7 to be active, while relay 8 is shown to be inactive (while the relay in reality is powered and active).

So, all I did was using a command to activate relay 8. TerrariumPi activated all others, while no clock or environment is controlling those relays. After regular intervals, the relayboard can be heard to click while the relays are deactivated and activated quickly. 

[image]

theyosh, 2020-01-30:
Hmm, somehow the code was duplicated, and that could cause this? Not sure....

But I made some updates. Could you do a `git pull` and restart? And see if it is now working correctly?

I only have a 4 relay board. So, can't test 5-8 .

vinnio, 2020-01-30:
I have updated and restarted, but unfortunately, no change.. It is still switching relays 7,6 and 5 on. 8 is still shown as deactivated, while it is really powered on.

theyosh, 2020-02-01:
He, ik zie NL tekst in plaatjes, dus we doen het ff in NL.

Dus schakelaars 1 t/m 4 doen het goed?
Als TerrariumPI aangeeft dat schakelaar 5,6,7 aan is zijn deze dan ook aan op de schakelaar zelf? Ik heb ledjes per schakelaar, en kan dus zien of deze aan zijn.

Want als het blijkt dat de status verkeerd is, maar dat hij niet schakeld, dan is het minder erg dan wanneer er ook nog gechakeld wordt.

En wat geeft dan de drcontrol.py script aan als je daar de status opvraagt? `./drcontrol.py -r all -c state`
Want ik heb die code overgenomen. Dan zou die ook verkeerd moeten weergeven.

vinnio, 2020-02-01:
Inderdaad, nederlands gaat net iets makkelijker :)
Ik ben even aan het rondkijken geweest en ik denk dat het probleem zit in de drcontrol. Bij de documentatie van drcontrol staat een probleem genoemd met 8-relay boards. 
https://code.google.com/archive/p/drcontrol/issues
Issues 1, 2 en 5 zijn eigenlijk, denk ik, het probleem. 
`./drcontrol.py -r all -c state` geeft bijvoorbeeld maar 7 relays, en nummer 5 wordt niet getoond. Met de oplossing uit issue 2 heb ik dat aangepast. Maar relay 5 wordt niet goed weergegeven. Bijgevoegd even de outputs van de commando's voor elk afzonderlijke relay.

1 tot en met 4 werken wel goed.
Met de schakelaars-pagina de schakelaar aanzetten gaat ook goed voor relays 1 tot en met 4. Voor de anderen ontstaat er weer hetzelfde probleem.

Resultaat van klikken op schakelaar 8 (LED bij 8 aan, Terrariumpi geeft 8 aan):
`root@raspberrypi:/home/pi/drcontrol# ./drcontrol.py -d DAE001XA -r all -c state`
`OFF`
`OFF`
`OFF`
`OFF`
`OFF`
`OFF`
`ON`
`OFF`

En na enkele seconden, na een verversing door Terrariumpi (LED bij 7 aan, Terrariumpi geeft 7 aan)::
`root@raspberrypi:/home/pi/drcontrol# ./drcontrol.py -d DAE001XA -r all -c state`
`OFF`
`OFF`
`OFF`
`OFF`
`OFF`
`ON`
`OFF`
`OFF`

En na enkele seconden, na een verversing door Terrariumpi (LED bij 6 aan, Terrariumpi geeft 6 aan): :
`root@raspberrypi:/home/pi/drcontrol# ./drcontrol.py -d DAE001XA -r all -c state`
`OFF`
`OFF`
`OFF`
`OFF`
`ON`
`OFF`
`OFF`
`OFF`

En na enkele seconden, na een verversing door Terrariumpi (LED bij 5 aan, Terrariumpi geeft 5 aan):):
`root@raspberrypi:/home/pi/drcontrol# ./drcontrol.py -d DAE001XA -r all -c state`
`OFF`
`OFF`
`OFF`
`OFF`
`OFF`
`OFF`
`OFF`
`OFF`

Tot slot (nu is relay 5 LED uit) (LEDs uit, Terrariumpi geeft alles uit)::
`root@raspberrypi:/home/pi/drcontrol# ./drcontrol.py -d DAE001XA -r all -c state`
`OFF`
`OFF`
`OFF`
`OFF`
`OFF`
`OFF`
`OFF`
`OFF`

[OUTPUTdrcontrol.txt](https://github.com/theyosh/TerrariumPI/files/4143133/OUTPUTdrcontrol.txt)

theyosh, 2020-02-01:
Hmm, volgens die bug melding bij Google, moet er een compleet andere numering in komen. Zie het verschil:

https://github.com/theyosh/TerrariumPI/blob/master/terrariumSwitch.py#L260

vs
```python
address = { "1":"1", 
            "2":"2", 
            "3":"4", 
            "4":"8", 
            "5":"10",
            "6":"20", 
            "7":"40", 
            "8":"80", 
            "all":"FF" 
        }
```
Dus dan zou ik verwachten dat bij jou ook de eerste 4 schakelaars 'raar' doen. Maar dat is niet het geval begrijp ik?

En helaas, ze hebben dus niet de lijst voor het checken online gezet.

Dus, dan zouden we nu twee dingen moeten gaan checken. 

1. drcontrol gebruiken voor het **aan** zetten van je schakelaars. En zo controleren of je idd alle 8 aan en uit kunt doen met de drcontrol.py. Vergeet de status.
2. Als test 1 werkt, dan zouden we per switch moeten gaan puzzelen welke waardes op de regels https://github.com/theyosh/TerrariumPI/blob/master/terrariumSwitch.py#L307 t/m 322 moeten komen.

Aangezien ik alleen 4 relays heb, en deze lijken goed te werken, hoop ik dat deze niet aangepast hoeven te worden voor jouw versie.

Dus kun je met drcontrol.py een test doen, door alle schakelaars een keer aan en uit te doen. Om zo te controleren dat dat python script dat goed doet. Dan weet ik dat het schakelen goed werkt.
Dan blijft alleen het uitlezen van de status over als puzzle.

theyosh, 2020-02-01:
En als nog meer code opzoek, kom ik dit tegen:
https://github.com/petersbingham/dae-py-relay-controller/blob/master/dae_RelayBoard/dae_RelayBoard_D2XX.py#L127

Daar zie ik voor het uitlezen van de status, dat er verschillende code is tussen een 4 relais versie en een 8 relais versie.

Dus ik denk dat ik die code eens ga proberen. https://github.com/petersbingham/dae-py-relay-controller . Is ook redelijk up-2-date. Dus eens zien wat ik daarmee kan.

theyosh, 2020-03-11:
@vinnio Hmm, het duurde wat langer, maar ik ben bang dat ik hier zo geen oplossing voor heb.

Aangezien ik de code 4/5 jaar geleden op die drcontrol.py heb gebasseerd. En het kan zijn dat dit nu niet goed is. Echter, het is zolang geleden al gemaakt, dat ik eigenlijk niet zo happig ben om het in zijn geheel te gaan herschrijven.

Want dan zou ik die andere library moeten gaan gebruiken, maar dan ben ik bang dat de volgorde van de schakelaar wisseld.
```
---------
| 1 | 2 |
---------
| 3 | 4 |
---------
| 5 | 6 |
---------
| 7 | 8 |
---------
```

vs

```
---------
| 1 | 5 |
---------
| 2 | 6 |
---------
| 3 | 7 |
---------
| 4 | 8 |
---------
```

Dus ik ben bang dat voor iedereen die een 8 poort schakelaar heeft, de kans bestaat dat de numering anders wordt. En dat lijkt mij nou niet de bedoeling.

Dus ik ben eigenlijk geneigd om deze issue helaas te moeten terug draaien. Ook is er met de nieuwe Python library nodig aan te geven of je een 4 poorts of een 8 poorts hebt. Dat betekend dat deze informatie zal moeten worden ingevoerd. Of ik moet een losse 8 poorts switch gaan toevoegen. Maar dan zit ik met duplicaat code.

Ik zal nog een keer kijken of ik idd een 8 poort FDTI switch kan toevoegen. Dan zal het alleen voor mensen die met een 8 poort versie werken betekenen dat ze de switches moeten omzetten. (manueel in settings file, want interface kan dit niet)

theyosh, 2020-03-11:
Mag ik je wat anders vragen? Zie jij het zitten om zelf wat te testen in de code?

Als voor jou nu de eerste 4 switches goed werken, en er is interactie als je manueel wat doet, dan kan het misschien zijn, dat het gaat om maar 4 regels code die we moeten uitvogelen.

In de code op https://github.com/theyosh/TerrariumPI/blob/master/terrariumSwitch.py#L315

Zie een staan:

```
      if relay == "5":
        return testBit(data, 2)
      if relay == "6":
        return testBit(data, 4)
      if relay == "7":
        return testBit(data, 6)
      if relay == "8":
        return testBit(data, 8)
```
En je zei dat wanneer je switch 8 veranderde, mijn software switch 7 zag veranderen. Dus dan zou ik nu voorstellen om de regel met testBit(data,X) aan te passen naar:

```
      if relay == "5":
        return testBit(data, 2)
      if relay == "6":
        return testBit(data, 4)
      if relay == "7":
        return testBit(data, 8)
      if relay == "8":
        return testBit(data, 6)
```
Hier is dus de 6 en 8 omgedraaid. Dit omdat dat voor jou werkte. Als jij dan de overige drie kunt uitzoeken, misschien zijn we er dan alsnog. Ik kan het zelf niet testen. En ik zie het eigenlijk niet zitten om deze hele switch compleet te herschrijven... :(

vinnio, 2020-03-12:
Geen probleem. En het is inderdaad onbegonnen werk om de hele switch opnieuw op te bouwen. 
Ik zal kijken of je bovenstaande optie kan werken, ook voor de andere relay-switches. 
En anders kan ik natuurlijk de eerste 4 switches gebruiken voor mogelijke handmatige schakelingen (aangezien die wel goed geupdate worden), maar ik overleef het zonder ook wel. :)

theyosh, 2020-04-05:
Zijn er updates? Heb de werkende getallen gevonden?

theyosh, 2020-05-17:
Closing after no response for more then 2 months...

closed 2020-05-17 by theyosh