theyosh/TerrariumPI #350: VEML6075 can't figure out address?

xagesz, 2020-01-19:
I bought a VEML6075 from adafruit and I can't get the address right. I am trying to pull both UVA and UVB data. When i run the command for the i2c matrix in ssh i can see the sensor show up and act as though its there but in the webgui i just get errors trying any number of combination of things to get the right data. The wiki could use some info on basics for a couple of simple things to help out. DO love this piece of software though.

theyosh, 2020-01-19:
Hi,

as far as I know, is the VEML6075 sensor on I2C address 10: https://github.com/theyosh/TerrariumPI/wiki/Hardware#i2c-bus

So, you can enter this value in the address value in my software. So address value '10' should work. And then you can add a UVA, UVB and UV Index server (three times) with the same address.

But if you have done this, and it does not work, it could be a user right issue. Are you running it on a Raspberry PI zero? If so, then the software needs to run as user root.

xagesz, 2020-01-19:
That was the key. It is working now. Thank you.

theyosh, 2020-01-21:
Was it the address value of 10 or that you have to run it as root?

xagesz, 2020-01-21:
Yes. It was the address needing to be 10. I was trying, 0 , 1, I2C1, I2C0, 10x0, 0x10 but never tried 10.

theyosh, 2020-01-26:
Thanks for the feedback

closed 2020-01-26 by theyosh