theyosh/TerrariumPI #351: DHT22 Alarm min/max settings reverting on reboot.

gale-force, 2020-01-20:
On a system reboot (from system settings actions) the the dht22 alarm min/max settings are changing to (possibly?) default values.

Limits, differences, names, addresses all persisting as configured.

Example: cool end temperature sensor alarm min is set to 18, on a reboot this changes to 14.

theyosh, 2020-01-21:
Hi,

could it be that you have specified a value in the environment settings for humidity or temperature at the field 'Day/night difference'? 

If so, then when you restart or reboot during night hours, the values can shift/change. So that should be avoided. This is something that is hard to fix. So reboots should only be done during daytime hours.

gale-force, 2020-01-21:
I have, yes, thats it.

Thanks!

closed 2020-01-21 by gale-force