theyosh/TerrariumPI #352: RF

polabear33, 2020-01-25:
I see you have compatibility with Energenie RF. Does this work with regular RF? I have been using rpi-rf from https://pypi.org/project/rpi-rf/

Am I able to set up RF through Terrarium PI? if so, how can I do it?

OvidiuTa, 2020-01-25:
Hello,
I am interested also to use this option, I am trying now with https://github.com/ninjablocks/433Utils/
The only way found was to use the Custom Script for the Switch, but it is not so easy because my python script need to write the state and have it persistent in a file. Also I found that from time to time I need to manually sync the status in the file, with the switch in Terranium and also with the physical plug. This is not nice :)
If someone is having an idea, for example to read somehow the status fro the software and sync auto will be nice.
Thanks!

theyosh, 2020-01-26:
Hi @polabear33 @OvidiuTa,

for both, I will not include software to support RF other then the Energenie. As I have a different RF device http://www.rfxcom.com/epages/78165469.sf/en_GB/?ObjectPath=/Shops/78165469/Products/18103 which is using different libraries. So, then I have to support multiple types of RF senders/receivers. Which is a bit hard.

Then, in order to toggle a power switch, you have to know hardware types, transmission types etc. All kind of technical data. There are tutorials to get those information by listening to commands of a remote. So that you can 'replay' those. I have done this in the past.

But this is rather difficult to make in my interface. As you have either to specify HEX commands for on and off. And I do not know how to use sensors. As that is also possible.

So, the only way to use this, is by making custom scripts, and use those scripts in my software. An example script: https://github.com/theyosh/TerrariumPI/blob/master/contrib/external_switch.py

For reading out data, it is a bit harder. Then the RF needs to be listening all the time. For that I cannot give any example.

theyosh, 2020-01-26:
@polabear33 Could you post your code that you have with the rpi-rf package. Maybe I can help you out in your case.

theyosh, 2020-01-26:
@OvidiuTa what kind of hardware do you use? Does a command like: './codesend ' work?

OvidiuTa, 2020-01-26:
> @OvidiuTa what kind of hardware do you use? Does a command like: './codesend ' work?

@theyosh I am using a Brennensstuhl Socket like [this](https://www.amazon.de/Brennenstuhl-Funkschalt-Set-Funksteckdosen-Funksteckdose-Kindersicherung/dp/B07CGBRS7T/ref=sr_1_fkmr0_1?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=socket+brennenstuhl+raspberry&qid=1580067500&sr=8-1-fkmr0). Yes, I use the ./send command inside my python script. I will take a look to your example, but I suppose that the state is not saved right?

OvidiuTa, 2020-01-26:
@theyosh I amplanning to buy also a Sonoff S20 and then to flash it via Tasmota. I see is supported from you sfotware.. I hope will be easier and more stable.

polabear33, 2020-01-26:
My code is very simple. It was originally for a program I wrote that would turn the fans on if temperature or humidity got too high. and turn it off if it was too low. It's definitely not pretty but it worked. 

I pulled out the applicable code, hopefully it makes sense
`from rpi_rf import RFDevice

#RF Setup
rfdevice = RFDevice(18)
rfdevice.enable_tx()
pulseLength = 167

#Thresholds
TempHigh = 88
TempLow = 83
HumHigh = 90
HumLow = 75

if AverageTemp < TempLow:
    switchCode = 5526988
    rfdevice.tx_code(switchCode, 1, pulseLength)
elif AverageTemp > TempHigh:
    switchCode = 5526979
    rfdevice.tx_code(switchCode, 1, pulseLength)
else:
    switchCode = 5526988
    rfdevice.tx_code(switchCode, 1, pulseLength)

if AverageHum < HumLow:
    switchCode = 5526835
    rfdevice.tx_code(switchCode, 1, pulseLength)
elif AverageHum > HumHigh:
    switchCode = 5526844
    rfdevice.tx_code(switchCode, 1, pulseLength)
else:
    switchCode = 5526844
    rfdevice.tx_code(switchCode, 1, pulseLength)`

theyosh, 2020-01-26:
@polabear33 I see the logic. So this is what I came up with...

```python
#!/usr/bin/env python
import sys
from rpi_rf import RFDevice

# This is just an example script to test the functionality for TerrariumPI. It is also a reference for implementing your own scripts

if len(sys.argv) == 1:
  raise SyntaxError('No arguments specified')

try:
  value = int(sys.argv[1])
except Exception as ex:
  raise ValueError('Invalid input value. Should be a number between 0 and 100 including')

if not (0 <= value <= 100):
  raise ValueError('Invalid input value. Should be a number between 0 and 100 including')

if value == 0:
  print('Toggle power switch to state OFF')
  # Call your code here....
  # Set the off command code for this power switch
  switchCode = 5526988
  
else:
  print('Toggle power switch to state ON or dimming value: {}'.format(value))
  # Call your code here....
  # Set the on command code for this power switch
  switchCode = 5526979

# Here we load the RF Device, because the values are valid
rfdevice = RFDevice(18)
rfdevice.enable_tx()
pulseLength = 167
rfdevice.tx_code(switchCode, 1, pulseLength)
```

You have to make as many scripts as you have switches. If you are handy, you could work with symlinks and 'basename' tricks in code. But for now, just create a single script per power switch.

Make sure the right python version is used in the first line. Either use python2 or python3 depending on your setup.

The logic to toggle is done in my software, so the values checks are not needed anymore.

Finally make this script executable 'chmod +x [scripts.py]' so that ./[script.py] 1 will power on the switch and with the parameter '0' off. This will then also done with my software when it needs to trigger/change the power switch

theyosh, 2020-01-26:
@OvidiuTa for you I think the above is the same, as you have to use your ./send command in your script.

About remembering the state, that is my software is doing. So you do not need to know the state of the power switch. My software will take care of that.

But, if you manually change the state outside my software, I cannot see that, and will not update the interface.

polabear33, 2020-01-27:
Awesome, I appreciate the help. How do I implement the code? in the "#call your code here...." Am I entering code into that? To be honest, my code was not that advanced and nor are my python skills. I can follow along, but the sys.argv coding is beyond me.

theyosh, 2020-01-30:
@polabear33 you can just copy paste my code in a new python file. Lets call it `switch_heater.py` and store it to make it easy in `/home/pi'

`nano /home/pi/switch_heater.py` and paste the code.

Then make the file executable: `chmod +x /home/pi/switch_heater.py`

Then when you run: `/home/pi/switch_heater.py 1` the power switch should toggle on. And with `/home/pi/switch_heater.py 0` it should toggle off. If that is the case, you can then use the custom script switch with '/home/pi/switch_heater.py' as address. Then you can use TerrariumPI to control that RF Remote switch.

And this can be done per RF remote switch. Make sure all the custom scripts do have a unique name.

theyosh, 2020-03-14:
Any updates? I can't do much more. Integrating custom hardware setups do require some coding

theyosh, 2020-04-05:
Closing due to no response in 2 months

closed 2020-04-05 by theyosh

polabear33, 2020-08-02:
I took some time away from this project and just now jumped back into it. I was able to write out the scripts I needed and confirmed they worked using the variables of switch_heater.py 1 and 0. However, when I plug them into the Terrarium PI, they don't seem to work. 

Here are the settings for it. When I click the power on/off buttons for them, it does nothing. The color does change and acts like it works, but there is no change on the switches.

[image]

Any help is greatly appreciated.

Thank you!

theyosh, 2020-08-04:
Hi,

are the scripts executable? `chmod +x [script_file]`

And do you have a shebang `#!/usr/bin/env python` in top of your script?

And what happens when you run as normal pi user: `/home/pi/Software/switch_heater.py 100` this should be the command that is run by my software.

Also, could you provide a debug screen: https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi

polabear33, 2020-08-05:
The scripts are executable

The shebang is there at the top on both. 

I get this error: /usr/bin/env: ‘python\r’: No such file or directory
-not sure what this means, looking into a fix.

Nothing unexpected in this image. I let it run a couple minutes and saw the same things. Temperatures are out of range because no heat or humidity is running. Looks like a couple sensors i'm having issues with. but nothing else out of the ordinary.

[image]

polabear33, 2020-08-05:
I was able to resolve the issue. I needed to use dos2unix to change my line endings. Then I also needed to change the shebang to python3. Now the buttons are working as intended

polabear33, 2020-08-06:
Alright, the buttons and all work. However the environment doesn't seem to be working properly. It's not turning on any of the switches. 

[image]
[image]
[image]
[image]
[image]

theyosh, 2020-08-08:
As it looks like in the logs, you have 'failing' sensors due to incorrect settings. You have set the limit_min and max wrong, so that current measured values are not within that range.

So, make sure that all sensors are working correctly, which means no error badges at the sensor pages.

The idea of limit min and max, is that some sensors are not stable, and can give invalid values. Like more then 100% humidity. That is not allowed. The fields limit min and max will control this. So set those values first on 0 and 100.

You can control the triggers of the environment with the alarm min and max values.

So, first make sure all your sensors have the correct limit min and max values. Then fine tune the it with the alarm min and max values for your environment setup. I thinks this will solve it.

polabear33, 2020-08-08:
Perfect! That worked! 

Thank you for all your help!