theyosh/TerrariumPI #353: Offline system

5point0, 2020-02-07:
Hi @theyosh,

So my internet is off at the moment, some link issue with my service provider. Anyways, I noticed that TerrariumPI won’t start up without internet.  

I have attached the error I’m getting.
[image]

theyosh, 2020-02-07:
Ah crap... did not test that....

And you are now offline, so an update through git is not working. So I have a manual fix for you.

Edit the file 'terrariumEngine.py' and go to line 175 and add 1 new line, and add two spaces in front of the three lines below. It should go from:

```python
version_data = terrariumUtils.get_remote_data('https://api.github.com/repos/theyosh/TerrariumPI/releases/latest',json=True)
self.update_version = version_data['tag_name']
self.update_available = int(self.update_version.replace('.','')) > int(self.current_version.replace('.',''))
self.update_last_check = datetime.datetime.now()
```

to
```python
version_data = terrariumUtils.get_remote_data('https://api.github.com/repos/theyosh/TerrariumPI/releases/latest',json=True)
if version_data is not None:
  self.update_version = version_data['tag_name']
  self.update_available = int(self.update_version.replace('.','')) > int(self.current_version.replace('.',''))
  self.update_last_check = datetime.datetime.now()
```
That should make it start up again. But then it could also have problems getting the weather data. And if you have you lights running based on sun set and rise, they will go to 8:00 and 20:00 due to missing weather data..

5point0, 2020-02-09:
Thanks @theyosh!

Internet is backup and running.

closed 2020-02-09 by 5point0