theyosh/TerrariumPI #354: request new hardware AC dimmer robotdyn

jeff26300, 2020-02-09:
Hello,

I would like to use for my heating lamp this cheap AC dimmer from Robotdyn: https://robotdyn.com/ac-light-dimmer-module-1-channel-3-3v-5v-logic-ac-50-60hz-220v-110v.html
I installed it with the PWN connected to GPIO12 (PWM0) of my Pi Zero but the dimmer function doesn't work, only on and off with 0 and 100%, so at least TerrariumPI is communicating with the dimmer.
Is this type of dimmer supported by TerrariumPI?

Thanks again for your support!

AndrejWeber, 2020-02-10:
Hey Jeff,
The AC-dimmer you found will not work without additional hardware. The shop even tells you, that you need it.
There is hardware, that you can connect directly to raspberry pi, but this hardware is a little bit more expensive, like this, theyosh is using for his dimmer.

theyosh, 2020-02-10:
hehe, somehow as soon someone starts about PWM or dimmers, @AndrejWeber comes around ;) Thanks a lot!

And yes, as far I understand, it is not a PWM dimmer, but a PSM. That works different I guess.

So, I use still with great pleasure: https://www.tindie.com/products/nEXT_EVO1/universal-ac-mains-dimmer-mpdmv41/

Shipping is a bit expensive, but that works like a charm.

So I am afraid the Robotdyn will not be supported by my.

jeff26300, 2020-02-11:
Thank you @AndrejWeber and @theyosh ! I indeed saw the MPDMv41 but the shipping costs are as expensive as the item itself :open_mouth: I will for now make a try with the on-off relay I already installed and place a manual dimmer since this is for the basking spot of the geckos only.
I am closing this issue

closed 2020-02-11 by jeff26300