theyosh/TerrariumPI #356: Error: Too many open files

swekley, 2020-02-26:
After doing a git pull around 2 weeks ago, TerrariumPi crashes with the following error

```
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/gevent/baseserver.py", line 199, in _do_read
  File "/usr/local/lib/python2.7/dist-packages/gevent/server.py", line 178, in do_read
  File "/usr/local/lib/python2.7/dist-packages/gevent/_socket2.py", line 190, in accept
error: [Errno 24] Too many open files
2020-02-26T15:32:17Z <WSGIServer at 0x642e5770 fileno=21 address=('::', 8090, 0, 0)> failed with error
```
When I googled "too many open files" I can only find solutions by expanding the open files limit using "ulimit -n <amount>", which i set at 10240 instead of 1024. Problem I still have using that is when TerrariumPi crashes, "ulimit -a" shows that the limit is back at 1024. I assume this is a bug in TerrariumPi as this never happened before the update. I think it's opening a file or connection without ever closing it.

swekley, 2020-02-27:
I managed to set the open files limit permanently to 65535, but TerrariumPi still crashes with the same error

theyosh, 2020-02-27:
Hmm, I am afraid this is not related to my software. As I do not have this issue, and I am running also on python2.7 for about 10 days now.

I have only about 400 open files as user PI. So I never get to the current max of 1024 for the Pi user.

I think the right approach now is to make sure all your package are up to date. So I would rerun the installer and then reboot.

Also, I do not have control about the gevent package. So I think this is something else.

Is your setup also accessible through the internet? Or do you have a connection open with your browser. As gevent is only used for webtraffic.

theyosh, 2020-02-27:
Also, you could install the package lsof: `sudo apt install lsof` and then run as root: `lsof -u pi`

This will give a very long list of open files by the user Pi. I am not sure if it helps to post it here, as I expect in your case more then 1024 lines.... which is a bit to much to post here.

swekley, 2020-02-27:
I can't access TerrariumPi when it gives me the too many open files, that error repeats every second in console btw.
The results of `lsof -u pi` are here: https://pastebin.com/9t8kiWup
I think it has something to do with i2c because `python   15253   pi  982u      CHR       89,1       0t0       9286 /dev/i2c-1` is in there 1004 times

theyosh, 2020-02-27:
Hmm then I would think this is a hardware failure.

Have you tried to reboot the Pi? Does that help? Also, I know that some hardware needs a power down when they have problems. So, it could help to power off your pi, and start it after a few seconds (10 or so). A single reboot will leave the power on the hardware.

swekley, 2020-02-27:
I'm having this error for about 2 weeks and even switched the powersupply, which works for only a couple of hours. I am using a USB stick instead of an SD card to boot the Pi, could that maybe be the cause?

theyosh, 2020-02-27:
Or, can you give me a list of hardware you are using that works through the I2C bus?

I also have some, but maybe you have a sensor that is maybe not working correctly due to my code. So if you can give me a list of sensors you are using, I can look at the code to see if I did something strange for one of your sensors

theyosh, 2020-02-27:
I do not think using USB is a problem. I really think that one of your sensors has died.

If you reboot, can you see the logs? Is there a sensor failing?

swekley, 2020-02-27:
Well the mh_z19 didn't work in TerrariumPi but works in the shell so I haven't configured that sensor anymore, but that's not i2c.
HTU21D - https://github.com/theyosh/TerrariumPI/issues/314 it worked for a week then it gave the exact same error
mlx90614 - works fine, never gave any errors

swekley, 2020-02-27:
I'll email you the SSH login

closed 2020-03-04 by swekley