theyosh/TerrariumPI #357: Rest API Help

Ultameciia, 2020-02-27:
It's less of an issue with TerrariumPi and more asking for help. 

I'm looking to find a way to leverage the API to pull sensor data from multiple terrariumPi environments and bring them into one dashboard for monitoring. This way I can display info from all my enclosures on a TV I've hung up on the wall in the reptile room. 

My coding experience is fairly limited, so I'm trying to find a solution that doesn't require me to write anything. Does anyone have any ideas or has done this?

theyosh, 2020-03-01:
Hmm,

that will be a bit hard. There is no 'simple' solution for this. The first problem is, either you have to run your own software internally, or every raspberry has to have a direct internet connection.

So a few solutions I can think off are:
 - Use a extra raspberry pi, and install my software. Then you can add external sensors, which are all your sensors of the other PIs. This can also done for the power switches. Toggling will not work, but they will show the current state per switch. But that is all. There is no fancy graphing.
- https://www.geckoboard.com/ But that is not free. And also, you need either to open firewall ports to make all your PIs directly accessible. Or you have to add scripts on all PIs to post data to online.
- https://freeboard.io/ same as above
- https://corlysis.com/ same as above.

So, I don't think there is an easy solution for this. And it will involve make some scripts. I have seen some simple examples, so I think it is not that hard.
A special as you have running multiple TerrariumPIs that looks like you are somehow handy in it. So https://corlysis.com/ is free. And maybe that is suitable for you even it involves some coding

Ultameciia, 2020-03-01:
I'm a huge proponent of the KISS theory. Setting up a Pi that has all sensors added makes the most sense I think. I guess I missed that update! Is there a sensor limit? I've got 9 Pi's running a minumum of two sensors each. Also is there some documentation on how to setup external sensors?

Ultameciia, 2020-03-04:
Found the documentation, so this issue is solved. I'll close it.

closed 2020-03-04 by Ultameciia