theyosh/TerrariumPI #359: Remote Sensor Conversion Bug

Ultameciia, 2020-03-04:
Referencing issue #357 I've set up another Pi that's working as a main hub for my other sensors. I've stumbled across a bug however and I can replicate it. 

On system A I've got an SHT2XX sensor reporting accurately and the system is set to display units in Fahrenheit. System B is configured the same way but with a DHT11. 

System C has the sensors added as remote sensors and the system is also set to display units in Fahrenheit. However the temperature sensors incorrectly report that my enclosure is at 168F. (poor baked tokay).

However if I change the system to display temps in Celsius then the out put is correct for Farenheit. The json output for "current" reports in Fahrenheit so I assume there's a weird conversion issue? Maybe the remote sensor expects the json to output in Celsius and then tries to convert that value to Fahrenheit again?

Ultameciia, 2020-03-04:
Just to add, both the SHT2XX and DHT11 sensors give me both humidity and temperature data. Humidity percentages are accurate when added as remote sensors.

theyosh, 2020-03-06:
> However if I change the system to display temps in Celsius then the out put is correct for Farenheit. The json output for "current" reports in Fahrenheit so I assume there's a weird conversion [...]

And that is correct. I do expect Celsius values for all sensors. So in your case that will give problems. So all your Pi's should wordt in Celsius and only the main Pi should use Fahrenheit at the moment.

I have to think how we can change this. I think that means adding more configuration options to the sensor settings. Not sure right now how to do this. Let me think about it.

Ultameciia, 2020-03-06:
That's the solution I found, all my enclosure Pi's are set to Celsius currently so that my main hub reports in Fahrenheit. I did notice that the value for current in the json does specify Fahrenheit or Celsius. It'll display something like 74.983f in the json output. Maybe that could be a way to specify whether or not the celsius to fahrenheit coversion takes place? 
(I'm a network engineer, not a programmer so maybe that isn't helpful haha)

theyosh, 2020-03-06:
I have thought a lot of how to work with temperature values. I thought about an option to specify it by input. But then, most hardware sensors can measure in Celsius. So then I thought, that is more easy.

But did not think about your setup, that it will only work for Celsius. So sorry, I am still European ;)

But, I think what I can do, is add some logic to the API that it can convert it realtime based on url. So lets add '/fahrenheit' to the url, to get the values in Fahrenheit despite the system setting. And so also '/celsius' etc. Give me some time to fix this.

theyosh, 2020-03-11:
@Ultameciia I made an update, so you can now change the API temperature output based on the url.

So for example:
`http://terrariumpi:8090/api/sensors/3d367d2738c7963a1bd6eb64eb0a6274`
Will show the temperature values in the system set values, but
`http://terrariumpi:8090/api/sensors/3d367d2738c7963a1bd6eb64eb0a6274/[fahrenheit/celcius/kelvin]`
will show the temperature value in the selected unit. It will convert internally and takes in account what the system is running in.

So for you this should/would mean that you can run all your TerrariumPIs in Fahrenheit, and then with the central TerrariumPI you use the remote url to all your sensors with the addition: `/celsius`

And then your central TerrariumPI can just run in Fahrenheit. But we force to read all the temperatures as Celsius as the code expect.

Ultameciia, 2020-03-12:
Ah, nice! I'm guessing just git pull?

theyosh, 2020-03-12:
Yes. Do a `git pull` and restart

Ultameciia, 2020-03-14:
Sorry for the late response, work has been hellish lately. Got everyone up to date and it's working well! Thanks!

theyosh, 2020-03-26:
Still working correctly? Can we close this issue?

Ultameciia, 2020-03-26:
Yep it sure is, I meant to close it with my last response.

closed 2020-03-26 by Ultameciia