theyosh/TerrariumPI #360: [Suggestion] Add camera options to rpicam

swekley, 2020-03-05:
Hi,

I'm using an IR pi camera and because of that my images are pink.
To prevent that I used to add `-awb greyworld` to the raspimjpeg command but I can't find anything related to awb in the code except for the telegrambot.

Could you add this as an option in camera settings? Something like `IR cam: yes/no`

theyosh, 2020-03-06:
hmm, greyworld is not yet support by the Picamera module in Python:

https://github.com/waveform80/picamera/pull/576
https://github.com/raspberrypi/firmware/issues/1167#issuecomment-511798033

So I have to make a hack for that. And I will add a White balance option in the settings, so you can pick your own. But this will take some days to make.

theyosh, 2020-03-06:
Updated the code. Do a `git pull` and restart. Make sure you have cleared your browser cache.

You should have a new Webcam setting. Change it to your needs and see if it works. It is not much tested yet.

swekley, 2020-03-06:
Finally no more pink pictures.. Thankyou! It works perfectly fine.

closed 2020-03-06 by swekley

andyhimself, 2021-03-21:
> Finally no more pink pictures.. Thankyou! It works perfectly fine.

How did you do it?
Im completely new to rpi camera, im using a NoIR cam and i need this feature. Can you show a step-by-step guide for us? Thanks

swekley, 2021-03-21:
I don’t use TerrariumPI at the moment, but I believe it was an option when
you add the camera. Choose “grayworld”