theyosh/TerrariumPI #361: DS18B20 Temp Probes Error After Short Time

ItsCaptainMorgan, 2020-03-05:
I have been trying make my own PI controlled environment for my Royal Python for way too long and have lost the motivation unfortunately.   I love the layout of this project and decided to throw it on my PI with a 5in touch screen.  3 DS18B20 temp probes to get a gradient.  2 GPIO relays: one timer for lighting one controlled by single temp probe.  

While trying to get temperature controlled relays to work I discovered that my temp probes go into error after a short period of time and the sensor displays get stuck.  I also have to have my gpio controlled relay board set to gpio inverse.   Any advise on getting this to work???  See images.
[image]
[image]
[image]
[image]

Ultameciia, 2020-03-06:
If your temp probes aren't reporting it's most likely due to your limit mins and maxes. Your limits are between 20-45c but the probes are reporting 12c so since those values are outside the limit min and max they're being discarded. Open up those limits to say 5-50c and see if they start reporting properly.

theyosh, 2020-03-06:
@ItsCaptainMorgan this is correct. As @Ultameciia is explaining. Your measured values of 12 degrees are lower than the limits values.

The limits values are there to make sure that measurements are correct. Sometimes some sensors are broken, and give to low or to high values. Then they will be ignored by the limit fields.

The alarm fields are the values where actions needs to be done. Either manual or through the environment system.

so increase you limit values as @Ultameciia suggested and it should work.

@Ultameciia thanks for answering.

theyosh, 2020-03-11:
@ItsCaptainMorgan Have you solved this? Is this still an issue?

ItsCaptainMorgan, 2020-03-13:
@Ultameciia @theyosh    This seems to work!   Is there a way to enlarge the temperature reading font size?   I would like to hook this into my 7in touchscreen for the PI but the temp readings are almost impossible to read.

theyosh, 2020-03-14:
Could you make a screenshot and add some markers of what text you are mention? Most of the design is a template that has fixed font sizes. So it depends on what you want if I can add that.

Mostly I would say, make you own web interface with API calls. So you can create you own kind of dashboard.

ItsCaptainMorgan, 2020-03-17:
I thought it would be a template but worth the question.  Here are the main text(s) I am referring to.
[image]

theyosh, 2020-03-21:
Hmm, that would be creating another hack. I rather do not do that. As I am not a UX designer, I use a predefined template.

So, for now, I am afraid that I cannot make a variable font size for the graphs.

Also, I think if you are handy enough, to make your own interface with the API data that is available. https://github.com/theyosh/TerrariumPI/wiki/API

ItsCaptainMorgan, 2020-03-22:
Okay!   Thanks for the help!!  Seems to be working how I need for now.  Now for electronics enclosure.

closed 2020-03-22 by ItsCaptainMorgan