theyosh/TerrariumPI #363: some bugs

swekley, 2020-03-14:
Hi,

I just installed TerrariumPi on a fresh pi 4 and it started up with a traceback. Also on a pi 3b+ I'm already using I faced 2 problems. 

One is that I can't see the archived pictures at night that are taken every 5 minutes during the day.
The other one is at the environment page. When I try to set the timer for the lights, the submit button doesn't work. The pictures below show chrome's console and terrariumpi's log
[image]
[image]

And here's the traceback on the fresh install
[image]

theyosh, 2020-03-14:
Hi,
ok lets try to split some things up. I have tested my software without any settings, so it should run like a clean install.

Error 1 on the first image in green is fixed. So with a `git pull` should fix that.

The error about PWM is unknown to me and sounds like Rpi4 related. As I do not have yet tested a Rpi4, I cannot tell how to solve this. Maybe it is a bug related to Rpi4.

The errors in the last image are also unclear to me. Looks more like a hardware/OS issue, then my software. I have seen some of those messages as well when I use Python3. But those are gevent library related, so outside my scope. So, I do not have an answer for that. Maybe trying to running the installer once more? I have no clue.

More people are saying that temperatures are changing during reboots. I have tested this multiple times in the last years, and the do NOT change. I can never replicate it, and 90% is caused by other errors or strange setup. It is known not to change sensors during the 'night' period if you have temperate offset set. 

Your issue with the times is not clear. So there I need more info. Could you create a screenshot of what you are trying to do?

Also, I suggest to open more issues, with 1 problem per issue. As you can see that I have create a very long answer, which will get more and more complicated when you try to fix 4 issues in 1.

swekley, 2020-03-16:
> The error about PWM is unknown to me and sounds like Rpi4 related. As I do not have yet tested a Rpi4, I cannot tell how to solve this. Maybe it is a bug related to Rpi4.

I'm trying to use a pwm or signal controlled MOSFET to dim the 10v dim line (pwm/0-10v/resistance) of a Meanwell powersupply used in our lamps. Should I then use "pwm dimmer" in your software?
[Exact product](https://opencircuit.nl/Product/IRF520-mosfet-module)
[Datasheet](https://www.vishay.com/docs/91017/91017.pdf)
 
> More people are saying that temperatures are changing during reboots. I have tested this multiple times in the last years, and the do NOT change. I can never replicate it, and 90% is caused by other [...]

Isn't it kind of a bug that you should not change sensors at night if you use temperature offset?
Maybe it's better to set a day/night temperature value when installing a temperature sensor, where if you don't fill in the night temperature it would use the day temperature. 
>

> The errors in the last image are also unclear to me. Looks more like a hardware/OS issue, then my software. I have seen some of those messages as well when I use Python3. But those are gevent [...]

I reinstalled and I did not get the error in console anymore when I try to set the timer, but unfortunately 'submit' still doesn't work. see screenshot below

> Your issue with the times is not clear. So there I need more info. Could you create a screenshot of what you are trying to do?
>

[image]

> Also, I suggest to open more issues, with 1 problem per issue. As you can see that I have create a very long answer, which will get more and more complicated when you try to fix 4 issues in 1.
>

I'll do this from now on.

theyosh, 2020-03-22:
> I'm trying to use a pwm or signal controlled MOSFET to dim the 10v dim line (pwm/0-10v/resistance) of a Meanwell powersupply used in our lamps. Should I then use "pwm dimmer" in your software? [...]
Not sure, but have you entered the pin number 4 or 7? Because number 4 is a 5V power pin. So I think you should use number 7 for the dimmer. I use physical pin numbers: https://pinout.xyz/#
And also, it is a 10V dimmer, which I expect to be DC. So you should then use DC dimmer. But I do not have much experience with this. 
So I you have entered number 4, then change that first to number 7 and try again. If that does not help, try the hard ware setting DC dimmer.

And if nothing is working, I can add a new dimmer type, because I have seen an example which could work: https://github.com/DrLex0/MightyVariableFan/blob/master/pi_files/pwm_server.py#L97

swekley, 2020-03-23:
I tried pin 12, 19, 21, 24, 26, 32, 33, 38 and now also 7 and 11 but nothing works, even with the dc dimmer. Should the dimmer be set to manual mode to test this? I've tried both but I'm kinda confused about the difference between manual and not.

About the temperature problem, i've now added a humidifier and those values also change when I  only change the light timer. Without any errors tho so I have no idea why it does that.

theyosh, 2020-04-16:
Sorry for the delay. I made an update trying to support the IRF520 Mosfet Dimmer. So do a `git pull` and restart. It could be that that you need to rerun the installer.

But then, I had problems with Python2 and the latest gevent package. So if you rerun the installer, you have to downgrade gevent manual: `sudo pip2 install -u gevent==1.4.0`
If you run it already on python3, just rerun the installer and you should be good to go.

Make sure you clear your browser cache. A new power switch is available. Use the physical GPIO pin which it is connected to. And try it out.

theyosh, 2020-04-18:
I just ordered some IRF520 dimmers. So I can test them myself. And I do have an actual usage for it.

swekley, 2020-04-19:
I did a git pull and then reinstalled, but it's not working on python 3. The options like dimmer duration are not showing up. 
[image]
[image]

theyosh, 2020-04-22:
Updated the code. For me the dimmer is working for a 24V fan. Not sure if it will work for you, due to the selected dimming frequency settings. Do a git pull and restart. Also the interface issues should be fixed. Make sure you clear the cache

theyosh, 2020-04-22:
You need to use one of the following GPIO pins: 32, 33, 35 en 12. These are the Pi hardware PWM pins, and I like / want to support only hardware PWM. Reduces some load on the Pi. http://abyz.me.uk/rpi/pigpio/python.html#hardware_PWM (pin numbers are in BMC there)

theyosh, 2020-05-17:
Ok, are the still issues here? Because I would like to close this issue. If there issues, please open a new issue. And please one issue per issue.

theyosh, 2020-05-20:
Closing due to no response for more then two months

closed 2020-05-20 by theyosh