theyosh/TerrariumPI #364: Raspberry Pi Zero Root Needed for GPIO Access

Ultameciia, 2020-03-16:
This is an old issue, and definitely more of a wishlist type thing. But I've recently started using raspberry pi zero w's to reduce cost (i've got tons of enclosures ha). While running as root allows dht22's to work, I'm really confused as to why this is the case. My ds18b20's work fine as the Pi user, but i'm sure that's due to the fact they're one wire. 

Like I said before it's a wishlist thing, but could a second look into why root is needed be something you've got a tiny bit of spare time for? I know you're super busy and it's relatively trivial if the system isn't public facing since we've got a solution for it.

theyosh, 2020-03-26:
Hi,

I have done this multiple times. And I can't fix it, as I do not use RPI Zero for this.

I think this is something that is outside my reach to fix. I add there fore a known issue: https://github.com/theyosh/TerrariumPI/wiki/Known-issues#raspberry-pi-zero

Sorry, but for now I do not have any solution to fix this.

Ultameciia, 2020-03-26:
I'll keep digging into it myself and hopefully share some info with you later. I originally thought it was just the fact the gpio group wasn't available to the user Pi, but other scripts can access the gpio pins as the Pi user. I'm almost betting it's related to the adafruit library as that's the only one I've tested so far with this issue. 

But I think adding it to known issues is a good work around for now, they seem to be getting more popular so I'm sure others have ran into this as well!

closed 2020-04-01 by theyosh