theyosh/TerrariumPI #365: Problems with Sonoff S20

duenni, 2020-03-18:
Hi,

lately I get some errors related to my Sonoff S20. This is in the log:

```
2020-03-18 13:44:02,326 - INFO    - terrariumEnvironment - Update done in 0.00197 seconds. Waiting for 14.99803 seconds for next update
TypeError: 'NoneType' object has no attribute '__getitem__'
return terrariumPowerSwitch.ON if terrariumUtils.is_true(state['POWER']) else terrariumPowerSwitch.OFF
File "/home/pi/TerrariumPI/terrariumSwitch.py", line 656, in get_hardware_state
data = self.get_hardware_state()
File "/home/pi/TerrariumPI/terrariumSwitch.py", line 220, in update
self.power_switches[power_switch_id].update()
File "/home/pi/TerrariumPI/terrariumEngine.py", line 626, in __engine_loop
Traceback (most recent call last):
2020-03-18 13:43:52,026 - ERROR   - terrariumEngine      - Engine loop: Power switch has problems: 'NoneType' object has no attribute '__getitem__'
```

I had Terrarium Pi 3.9.6 running when my SD card got corrupted. I did a fresh install with the latest version and copied back settings.cfg and notifications.cfg from backup. System works fine but in 2 days I got 3 warning mails with this error. The switches are working. Any ideas?

theyosh, 2020-03-21:
Hmm, but those are not all the time? If it happens in a low frequency, I think it has to do with the connection. As I do not have a sonoff I cannot debug it. So that will be hard to fix.
I see based on your code it is flashed to Tasmota. And I do see in the code that there is not a real check on the returning data.

So if you can confirm that this is only happening once or twice a week, I think I can make a fix for this. But if it happens once every minute, we have a bigger problem

duenni, 2020-03-21:
Hi,

thanks for your answer. Yes, those are flashed with Tasmota. Well, it happens 3-4 times a day at random and it did not happen before with the old version (3.9.6). I did not change anything besides doing a new install. The devices involved stay in the same place, same hardware, same Tasmota version.

theyosh, 2020-03-21:
Hmm, I have no idea. I think there were a lot of small changes, but nothing regarding the Sonoff devices.

But I have updated the code, so that when it cannot read out the current state, it will return the old state. So that should at least fix the error. There will be a warning in the logging that this has happened.

So do a `git pull` and restart. And see if you still have warnings, but the program/system should continue to run.

duenni, 2020-03-22:
Mhhh, the system is also somehwat sluggish which it wasn't before (web ui takes >20 sec to load graphs, ssh struggles) . Then I remembered that I disabled IPv6 in the old system which improved performance. Maybe this is related, I just disabled IPv6 again in Raspbian and will now wait if it happens again. The Tasmota devices are IPv4 only.

duenni, 2020-03-25:
Well, that was not it. Still receiving the errors. 
Your fix contains a small typo "retruning".
Another question: shouldn't we check how often this happens? If the state could not be read 5 times in a row then throw a system error or something like that?

OvidiuTa, 2020-03-31:
Hello,
I post also here because I am not able to add a Sonoff socket. It is flashed with Tasmota 8.1.0.
I add it with the address: http://10.x.x.x and I get this error in logs:

2020-03-31 10:45:53,232 - ERROR   - terrariumSwitch      - Sonoff device is not connected. Cannot read power switch state

I need to say that the Pi and the Socket are in the same network.

@duenni @theyosh Could you please give me a hint what the problem can be?

theyosh, 2020-04-01:
> Mhhh, the system is also somehwat sluggish which it wasn't before (web ui takes >20 sec to load graphs, ssh struggles) . Then I remembered that I disabled IPv6 in the old system which improved [...]

I also have some sluggish connection. But I only see it when I connect from another wifi to my pi over wifi. When I connect from a wired device to my Pi, I do not experience that sluggishness...

But then again, when I look at some of my graphs, I see some more disk io (reading....) and some more CPU load. https://theyosh.nl/munin/theyosh.nl/terrariumpi.theyosh.nl/cpuspeed.html and https://theyosh.nl/munin/theyosh.nl/terrariumpi.theyosh.nl/diskstats_iops/mmcblk0.html

But I have looked at my commits in January 2020, and I can't find anything. As I only had translation commits. So I can't say if it is my code. I can also be that the OS has changed something... and I have a pretty old SD card, which is getting slow...

So at the moment, I do get the feeling my setup is getting slower, I do have to take in account that my sqlite DB is almost 5GB already, and my SD card is at least 3 years old :(

At the moment, I do not have a clue where to look to solve this....

theyosh, 2020-04-05:
At the moment I am using also a Sonoff and I do see some connection issues. But they are very little. 

So I made a small update that it will try to reconnect for 5 times when it could not load the hardware the first time. I hope this will solve most of this issue.

The only thing I cannot find/fix is the sluggishness. I can't find it in my code. @duenni Did you reinstall with the same OS? Was you OS up to date before you updated? Could it be that it is OS related?

duenni, 2020-04-05:
Hi! Sluggishness is gone since I disabled IPv6 on the Pi. Maybe this is related to my internal network, I don't know.
Thanks for the fix, will try it out. 👍 
Well, everything works fine, I just get some mails a day because of the connection problems to the Sonoff devices. But they are on a timed schedule and switch on and off just fine.

theyosh, 2020-04-13:
Ok, I am using a Sonoff as well now, and for that last week zero issues. It can connect all the time. So I do not see any issues with the power switch. Are you using a hostname or did you enter an IP number in my software?

duenni, 2020-04-18:
I am using a hostname and did not get a mail since I use the latest version.

theyosh, 2020-05-01:
Ok, I think we are going to close this? I did made some code updates to handle connection errors. But I do not see them on my system. The only times I see them, is after a reboot of the Pi. Somehow, then it is not able to connect within the fist minute.... strange.... But those are rare cases, which I just say: DO NOT REBOOT :)

So closing this?

theyosh, 2020-05-02:
I am closing for now. Seems some kind of strange corner cases which do not happen that often.

closed 2020-05-02 by theyosh