theyosh/TerrariumPI #366: Sensor error after update

5point0, 2020-03-21:
Hi @theyosh 

I have done a git pull and installed but I get the following issue:

    2020-03-21 21:22:09,000 - INFO    - terrariumWrapper     - Starting TerrariumPI server at location: http://192.168.88.140:8090 ...
    2020-03-21 21:22:18,615 - WARNING - terrariumSensor      - OWFS file system is not actve / installed on this device! If this is not correct, try 'i2cdetect -y 1' to see if device is connected.
    2020-03-21 21:22:23,675 - WARNING - terrariumBluetoothSensor - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
    2020-03-21 21:22:35,122 - WARNING - terrariumSensor      - Measured value 482.0ppm from k30co2 sensor 'co2' is outside valid range 0.00ppm - 100.00ppm in 1.50406 seconds.
    2020-03-21 21:22:56,000 - INFO    - terrariumWebserver   - Running webserver at :::8090
    2020-03-21 21:22:58,591 - ERROR   - terrariumEngine      - Engine loop: Sensor has problems: get_data() got an unexpected keyword argument 'temperature_type'
    Traceback (most recent call last):
      File "/home/pi/TerrariumPI/terrariumEngine.py", line 618, in __engine_loop
    self.get_sensors([sensorid],socket=True)
      File "/home/pi/TerrariumPI/terrariumEngine.py", line 947, in get_sensors
    data.append(self.sensors[filtertype].get_data(temperature_type=temperature_type))
    TypeError: get_data() got an unexpected keyword argument 'temperature_type'
    2020-03-21 21:22:58,715 - ERROR   - terrariumEngine      - Engine loop: Sensor has problems: get_data() got an unexpected keyword argument 'temperature_type'
     Traceback (most recent call last):
      File "/home/pi/TerrariumPI/terrariumEngine.py", line 618, in __engine_loop
    self.get_sensors([sensorid],socket=True)
      File "/home/pi/TerrariumPI/terrariumEngine.py", line 947, in get_sensors
    data.append(self.sensors[filtertype].get_data(temperature_type=temperature_type))
    TypeError: get_data() got an unexpected keyword argument 'temperature_type'
    Traceback (most recent call last):
      File "src/gevent/greenlet.py", line 766, in gevent._greenlet.Greenlet.run
      File "/home/pi/TerrariumPI/terrariumEngine.py", line 652, in __engine_loop
    average_data = self.get_sensors(['average'])['sensors']
      File "/home/pi/TerrariumPI/terrariumEngine.py", line 954, in get_sensors
    data.append(self.sensors[sensorid].get_data(temperature_type=temperature_type))
    TypeError: get_data() got an unexpected keyword argument 'temperature_type'
    2020-03-21T19:23:00Z <Greenlet at 0xa2fcf420: <bound method terrariumEngine.__engine_loop of <terrariumEngine.terrariumEngine object at 0xa4c19f90>>> failed with TypeError

None of the sensors seem to update and I cannot access the sensor setting page or view the sensors.

theyosh, 2020-03-21:
Crap, my fault. I missed the MiFlora sensor in an update for #359 
Do another git pull and it should work again

theyosh, 2020-03-26:
Is it working for you? Can we close this issue?

5point0, 2020-03-30:
Hi @theyosh 

Sorry I had rolled back to a previous version by forking, I didn't know your reply would be so quick😉. 
I will try get things back to normal later today and let you know.

theyosh, 2020-04-05:
Any updates... ? I would like to close this issue, and for that I would really got a confirmation from you it is working.

5point0, 2020-04-06:
I have moved back from the fork to the latest version and so far so good.
Thank you.

closed 2020-04-06 by 5point0