theyosh/TerrariumPI #367: Liquid Level Sensor

nastea1, 2020-03-24:
What's the possibility of adding a simple liquid level sensor?

http://www.cqrobot.wiki/index.php/Liquid_Level_Sensor

I currently use a python script and it uses it as a button, and it's defined as water.is_held , it returns dry and turns on a pump if it's not held, it reports as wet and therefore doesn't turn on the 5v Pump.

It's been modified from a soil sensor to say if soil is wet, don't water, if dry water. So this sensor sits in a tub of water and when water is not sensed, it fills it by firing a pump for 5s, then cycles around and if it's not full enough it continues.

theyosh, 2020-03-26:
Hi,

This is not really a sensor. This will only trigger on or off when it is in the water. So I can't measure anything. Then it is the question how to use this.

There is a sensor called 'YT-XX Digital'. You can use that sensor for this liquid sensor. As this sensor only reports 0 or 1 back. So the graph will only graph either 0 or 1.

But, if you measure this in a enclosed box, you can also do the following: https://user-images.githubusercontent.com/8080177/49337970-ef09ab80-f613-11e8-8187-e1e2c1985452.jpg

I also use a HC-SR04 to measure the distance between the water and to top of my water reservoir. When the distance is getting to much, you could add a trigger to start the water pump. This will also give you an option to specify the amount of water in liters that is still available. (Needs some math)

nastea1, 2020-03-27:
> Hi,
>
> This is not really a sensor. This will only trigger on or off when it is in the water. So I can't measure anything. Then it is the question how to use this.
>
> There is a sensor called 'YT-XX [...]

Thanks for the suggestions, I want to try the YT-XX Digital option because that's all this sensor does. It's very simple. However for the alarm though....when you're only getting 1 or 0.

nastea1, 2020-03-28:
> Hi,
>
> This is not really a sensor. This will only trigger on or off when it is in the water. So I can't measure anything. Then it is the question how to use this.
>
> There is a sensor called 'YT-XX [...]

So I've been testing around with a few different setups and I can't get this device to show any hits when it should be at least throwing a 1 every now and then. It does read current status as 0, but none of the graphs show any hits. Any suggestions? I guess I should just switch to volume or something then. It's difficult because the water reservoir that needs to be filled is open air and has a heating element under it to heat it and evaporate warm humidity. So in order to keep it full, the current sensor/switch detects when enough water has evaporated that the sensor becomes dry and then triggers the pump to fill it.

nastea1, 2020-03-30:
@theyosh the only thing I have been able to get this to trigger on is the door alarm....I wasn't able to get any graphs of any YTXXDigital because the sensor types don't just have a 1 or 0, they're in gallons, liters, ppm etc. 

But right now it shows door is open, when the water dish is full, just have to manually trigger the pump to fill it once it shows door is closed.

nastea1, 2020-03-30:
I basically just used the system to use them as either a conductivity sensor under YTXX. Once the Conductivity is detected at 100ms, it fires the pump, which fills the water from the 5gallon tank, it then detects that conductivity is 0ms once water hits the optical sensor. Secondarily it's set up to detect volume over 25gallons, using the same type of sensor and YTXX, once it detects that the sensor is over 25 gallons it triggers a high event (even though water is empty) and fires the relay to fill the other water storage. Once water hits the optical sensor it fires as 0 gallons, turning off the 'high event' and turning off the pump.

closed 2020-03-30 by nastea1