theyosh/TerrariumPI #369: Multiple or cycle timer?

nastea1, 2020-03-29:
Anyway to use the system as a controlled cycle timer? Say to turn on a device for x minutes at intervals or specific times etc? For example turn on a fan for 5 minutes once and hour. Or turn on a humidifier for 10 minutes every 15 minutes, etc.

nastea1, 2020-03-29:
Disregard I finally figured out the timing in the timers.

theyosh, 2020-04-01:
Yeah, it is possible what you want. And I do know the documentation is lacking a lot :( . But I glad you figured it out.

nastea1, 2020-04-01:
> Yeah, it is possible what you want. And I do know the documentation is lacking a lot :( . But I glad you figured it out.

@theyosh No problem, it's a great tool. You've done amazing! I'm using it to control a mushroom farm! So just different setup and learning curve is high, but it's wonderful!

nastea1, 2020-04-01:
Ooops didn't mean to open, just reply.

closed 2020-04-01 by nastea1